sinara-hw/Fastino #2: analog design notes

hartytp, 2019-08-29:
**noise density**
- for HOA2, 43Ca+, 1MHz secular frequency: 0.5nV/rtHz on all electrodes (uncorrelated) gives 10qps (quanta per second) heating rate
- heating rate is quadratic in voltage noise density (linear in PSD)
- heating rate scales as inverse of secular frequency
- we'd like to aim for <1nv/rtHz at our trap, assuming a passive multi-pole RC filter between Fastino and the trap
- assume we use a filter like https://ws680.nist.gov/publication/get_pdf.cfm?pub_id=912958 and assume Fastino has an output resistance of around 33 Ohm (to isolate the OpAmp from cable capacitance)
- assume we get around 35dB of filtering, that's a factor of ~50 in linear units. So, the noise density out of Fastino in the stop-band of the Fastino's filters needs to be <~50nV/rtHz, which isn't trivial but should be doable with careful design.

[image]

[image]

[image]

- rough ball-park of heating during a splitting operation. Suppose the secular frequency drops down to 300kHz for 30us and we want to limit axial heating to <1 phonon. That's a max heating rate of 33kHz at the lower frequency
- for 1MHz secular, that would be a noise density of `0.5*sqrt(33e3/10)=30nV/rtHz`. For 300kHz it needs to be 10 quanta/second.
- We'll likely be in the 50-100nV/rtHz, so heat by 25-100quanta over that splitting operation.
- noise sources:
  - AD5542A (DAC) Vn =12nV/rtHz after an AFE gain of 5 that's 60nV/rtHz. IIRC there isn't a significantly lower noise DAC around at this kind of BW, so this is a fairly hard limit. Only affects "in-band" noise though as removed by AFE filter stages
  - AD8620ARZ (AFE OpAmp) Vn=6nV/rtHz. IIRC the noise from the second OpAmp is still present after the filter with a gain of 2.5, so that's 15nV/rtHz which is fine.
  - OpAmp In negligible since JFET
  - [ ] to do: SPICE noise model for Stabilizer, look at relative contributions. Measure on current hardware to check reality matches the model

@dnadlinger IIRC you had a nice table of the noise for the various "fast DAC" cards around. Would you mind posting that for reference?

**slew rate**
- I'd like to be able to output FS sinusoids at a few hundred kHz without being slew-rate limited
- peak slew-rate limited is 10V*2*pi*300kHz=12.6V/us (in practice a little less is needed due to attenuation by the filter)
- the DAC has a slew rate of 17V/us after an AFE gain of 5 that's 85V/us
- the AD8620ARZ used on Stabilizer has a slew rate of min 40V/us
- so this should be completely fine!
**other comments**
- since we're using much more stable DACs than Zotino (and stable dividers) we don't need a TEC
- we'll need to do proper PI and thermal simulations to check how hot this gets

hartytp, 2019-08-29:
From Ludwig's thesis:

[image]

So the in-band noise would be about the same as the ETH cards.

dnadlinger, 2019-08-29:
Yep, the table I had was the one from Ludwig's thesis you posted.

hartytp, 2019-08-29:
- [ ] to check: can we switch to a 5V ref and 4xAFE gain for slightly lower noise?

hartytp, 2019-08-29:
> to check: can we switch to a 5V ref and 4xAFE gain for slightly lower noise?

AFAICT the DAC is fine with Vref=Vsupply=5.5V. IIRC the choice of 4.096V for Stabilizer was because of the ADCs, not the DACs.

For 5.5Vref we'd get nearly a factor of 2 reduction in heating rate for low axial frequencies which is definitely worth having. Lowering the gain of the final stage also lowers the out-of-band noise floor so all around a good thing to do. But, needs double checking from the data sheet that this is allowed.

dnadlinger, 2019-08-30:
Regarding op-amp noise, the OPA827 is 4 nV/rtHz at 1 kHz, i.e. 33% lower. Slew rate needs a look, though.

hartytp, 2019-08-30:
> Regarding op-amp noise, the OPA827 is 4 nV/rtHz at 1 kHz, i.e. 33% lower than the AD8620, and cheaper as well. Slew rate needs a look, though.

IIRC I rejected that originally because I wanted a dual OpAmp.

Not sure if we have the board real-estate for singles. Since the DAC is 11nV/rtHz the OpAmp noise does alter the overall output noise that much, although lower cost is good...

hartytp, 2019-08-30:
> and cheaper as well.

Not really since you'd need two of them and only one OPA8620

jordens, 2019-08-30:
The other relevant quantity here is the the output range. The +-10V is sounds about right for larger traps (i.e. HOA2 with 140µm characteristic distance). But smaller traps can live happily with less range. And even going to -4 to +8V would help with the noise. IME transport waveforms tend to be pretty asymmetric and need more positive headroom. At least in the traps that I built and worked with. Dunno for HOA2.

dnadlinger, 2019-08-30:
Ah, right, I only had a look at singles. I actually meant the new OPA828, though, but that is still only single, and not available in packages smaller than SOIC.  £3.51 in qty 1000 on Digi-Key, though, vs. £9.95 for the AD8620ARZ.

jordens, 2019-08-30:
And do check the quiescent current. The fancier opamps sometimes buy low noise with lots of quiescent bias current for internal stages.

gkasprow, 2019-08-30:
If we get rid of CMCs, we would spare some real estate for the second opamp.

hartytp, 2019-08-30:
> If we get rid of CMCs, we would spare some real estate for the second opamp.

Sure. All things equal, I'd still prefer to use dual OpAmps though and keep the layout compact.

> And do check the quiescent current. The fancier opamps sometimes buy low noise with lots of quiescent bias current for internal stages.

Yes, thermal/power issues are high on my list of concerns for this design.

dtcallcock, 2019-08-31:
> The +-10V is sounds about right for larger traps (i.e. HOA2 with 140µm characteristic distance).

HOA 2 is rated to +/-20V according to the [manual](https://prod.sandia.gov/techlib/access-control.cgi/2016/160796r.pdf).  Might be worth checking if users of overly-bulky ions like Yb+ actually make use of this capability? @jbqubit ? 

> And even going to -4 to +8V would help with the noise. IME transport waveforms tend to be pretty asymmetric and need more positive headroom.

OPA8620 is a 26V opamp so if we went this route perhaps we could make it so that one could switch between -4/+8 and -8/+16 (probably via population options)?

hartytp, 2019-08-31:
@jordens 
> The other relevant quantity here is the the output range. The +-10V is sounds about right for larger traps (i.e. HOA2 with 140µm characteristic distance). But smaller traps can live happily with [...]

@dtcallcock 
> HOA 2 is rated to +/-20V according to the manual. Might be worth checking if users of overly-bulky ions like Yb+ actually make use of this capability? @jbqubit ?
> OPA8620 is a 26V opamp so if we went this route perhaps we could make it so that one could switch between -4/+8 and -8/+16 (probably via population options)?

These are all valid points. Taking a step back for a moment, I think it's important to have a clear mission statement for any project to avoid a bloated, overly-complicated design.

My aim with Fastino is
(a) to create a general-purpose workhorse "Fast, low-noise Sinara DAC" (for some definition of those terms), which is something that's sorely lacking right now IMHO
(b) to make sure the design is generally the right kind of thing for ion transport experiments in terms of the basic parameters like sample rate, DAC noise, etc, but without trying overly hard to optimise it for any particular ion species/trap/experiment.

In particular, I'm working on the assumption that designing AFEs/filters is relatively easy and anyone who really cares about performance will probably customise the AFE design based on the precise waveforms they need to produce anyway. I definitely expect that at the very least, we will end up changing some RC values for the version we end up using in our experiments.

hartytp, 2019-08-31:
So, my preference is to keep it as a +-10V DAC with a 2-pole filter at something like 500kHz. That should work well for the majority of use-cases, and I don't think there is a perfect solution here that will work for everyone (across ion species, traps, experiments) without customisation.

hartytp, 2019-08-31:
For the OpAmp, my aims are:
- minimise degradation of the DAC noise inside the filter pass-band, and try and keep the out-of-band noise not too high (although, we will probably have ~1k=4nv/rtHz resistors in the feedback loop so no point overly pushing on this)
- don't be slew rate limited for the kind of signals we can produce
- ideally be a dual OpAmp to keep the design compact (this leaves room for things like extra resistors for asymmetric ranges in the future)
- be as cheap as possible
- have low quiescent current
- high-enough GBP for the filter to still have good performance at a few MHz

At some point next week, I'll do the usual parametric search of the main manufacturers and summarise our options.

hartytp, 2019-08-31:
On the subject of asymmetric voltage ranges:
- that's a little messy, as it means we no longer use the offsetting resistors in the DAC, so have to add our own
- this means either using an extra resistor ladder per channel (more space and money) or doing something like standard resistors with a sensible temp co and add a calibration ADC
- population options for this makes it even messier

That pushes me to think it's not a path to go down for the basic Fastino design. But it's definitely something users should consider implementing on their own Forks/customized versions.

dtcallcock, 2019-08-31:
> I don't think there is a perfect solution here that will work for everyone ... ion species

Yeah, and looking at the strike-throughs on the Sinara team page, I'm not sure there are even any Yb-ers left.

40V op-amps would clearly need a new AFE design.  Just wanted to point out an opportunity to flog more boards and prevent fragmentation if we went with @jordens asymmetric suggestion.

hartytp, 2019-08-31:
> Just wanted to point out an opportunity to flog more boards and prevent fragmentation if we went with @jordens asymmetric suggestion.

Maybe, but:
(a) There are now a lot of non-ion-trapper Sinara users who also want a Fastino-like design. My suspicion is that optimizing voltage ranges for ion transport will probably make it less ideal for other applications
(b) The +-10V ranges should work pretty well for everyone -- there isn't a huge decrease in noise by going from +-10V to [-4,+8V]. You'd only need that when you want to squeeze the last factor of a few in heating rate out, and by that point I suspect you'd want something custom-designed around your particular experiment anyway

In other words, I think "Fragmentation"/customisation at this level is inevitable and not a bad thing -- like the custom AFEs for Sayma. The key is to make a good base design that gets everyone off the ground well, and which can be tweaked easily for particular experiments.

hartytp, 2019-08-31:
@dtcallcock another example of this is bw. What BW do you go for? The Mainz group (based on the publication on their DAC) use a 2-pole active filter with a cut-off of around 50kHz to avoid heating during splitting when the trap frequency gets low. So, one approach one might want to take would be to bring the filter cut-off down, but keep the voltage range higher to allow pre-emphasis of higher frequency components. I wouldn't want to make the filter cut-offs that low for the "Base" version of Fastino, but it's something that users might well want to do on a Fork/Customization.

I really feel that the space of transport experiments is sufficiently large that we're not going to produce a single design that's optimal across all of it -- it you want to maximise performance you have to customise based on your particular experiment.

gkasprow, 2019-08-31:
We can make assembly variants where for every component we can define also a value. In this way, the client can simply order a production batch with specific output range generating his own BOM and PnP files from Altium directly.

hartytp, 2019-08-31:
For the rcs (filter bw and overall gain) that sounds like a good idea. But for things affecting topology (like asymmetric voltage ranges, which need extra dividers) it needs a little more thought and I don’t think I want to include it at least in the initial version

dnadlinger, 2019-08-31:
The OPA2210 might also just about work (good low-frequency noise, low current consumption, cheap, but slew rate is rather low and I haven't done the numbers what the bipolar input would do to the DAC precision).

Another thing not to forget when driving low-pass filters is the op-amp output current limit.

---

Generally, it seems like for a generic board, the easiest route would be to just put on a relatively high bandwidth filter that just makes sure any clock/update spurs at the target (max.) update rate are well suppressed. (Since that will be in the range of typical ion secular frequencies, does it make sense to include a notch?)

Even with a custom Fastino output stage, there would typically be a PCB close to the trap anyway to adapt to whatever obnoxious connector is used for the vacuum feedthrough, so why not just put any passive filters to optimize noise performance there?

hartytp, 2019-09-01:
> Generally, it seems like for a generic board, the easiest route would be to just put on a relatively high-bandwidth aliasing filter that just makes sure any clock/update spurs at the target (max.) update rate are well-suppressed.

> Even with a custom Fastino output stage, there would typically be a PCB close to the trap anyway to adapt to whatever obnoxious connector is used for the vacuum feedthrough, so why not just put any passive filters to optimize noise performance there?

That's basically what I'm proposing. Topologically, and ideally packages/layout-wise as well, I expect Fastino's AFE to be identical to Stabilizer (or, to put it another way, I designed Stabilizer's AFE as a prototype of Fastino).

All I want to do here is double check the OpAmp PN and RCs to check the design is roughly optimal in terms of noise/cost/power consumption. e.g. increasing the reference voltage to 5V and decreasing the AFE gain is an easy win (just change a few RC values) that gives us lower noise for free.

hartytp, 2019-09-01:
e.g. the 10k resistors in the output filter should be probably changed with 1k resistors after giving some brief thought to current consumption, otherwise they put a fairly high unnecessary floor on the noise outside of the filter's cut-off.

hartytp, 2019-09-01:
> The OPA2210 might also just about work (good low-frequency noise, low current consumption, cheap, but slew rate is rather low and I haven't done the numbers what the bipolar input would do to the DAC precision).

Slew rate looks a bit low (see above calculations). Also the current noise is a bit high given the DAC's high impedance at both the inverting and non-inverting pins. Not the end of the world, but a JFET amp would be better.

dnadlinger, 2019-09-02:
Re op amp choice, here's a table comparing some of the contenders, courtesy of the airport wifi: https://docs.google.com/spreadsheets/d/1cltL1o3Fes_si33FFXsfMfq_Re5cl5EuDKJbJWOh1aU

dnadlinger, 2019-09-02:
Nothing too surprising there. As expected, CMOS op-amps are cheap, but have worse 1/f noise. Bias current is listed at 25 °C; the usual JFET temperature caveats apply (i.e. numbers given are pretty useless, but I wasn't sure whether to include some high-beta bipolar parts). The ADA4625 looks like a very nice part, but the `-2` duals aren't available yet.

All in all, the AD8620 looks a bit overly expensive for what it offers.

hartytp, 2019-09-02:
Thanks! @dnadlinger that's really useful.

OPA2156 does look like a good contender. My only slight concern is the supply current. 32*24V*2*4.4mA=6.75W, which is a fair bit.

Sorting by cost, removing all parts with a supply current more than 4.4mA. Removing columns where all parts you listed are fine. The remainder of your table looks like this AFAICT

[image]

hartytp, 2019-09-02:
I'm not too worried about the larger offsets, since the [divider resistors](http://www.farnell.com/datasheets/1932989.pdf?_ga=2.39542499.197723750.1567441610-1538663289.1567441610) are only good to 5e-4 ratio = 10mV on 20V. 

So, I'd say the next step is to simulate the filter transfer function for a few different OpAmps from that selection and pick the lowest current consumption amp that does the job. Agreed?

dnadlinger, 2019-09-02:
> Agreed?

Yep; basically, if the OPA2197 is quick enough, there is little reason to look somewhere else. ADA4625 or OPA828 for faster JFET amps with significantly lower 1/f noise.

hartytp, 2019-09-02:
> ADA4625 and OPA828

I discarded those because they're quite pricey and thirsty so I don't really see the appeal over the significantly cheaper OPA2156.

> with significantly lower 1/f noise.

Is that a concern? LTC6655 is 0.25ppm 1/f noise, so 5uV on 20V output. All the amps you selected have low 1/f noise compared with that, so it's not worth worrying about, right?

dnadlinger, 2019-09-02:
> low 1/f noise […] not worth worrying about, right?

1.9 µV over 5.5 V is 0.4 ppm, plus if we really wanted, we could filter down the LTC6655 further using a quiet chopamp and capacitor-leakage-reduced RC filter setup. I don't think this is necessarily worth worrying about, though.

hartytp, 2019-09-02:
> 1.9 µV over 5.5 V is 0.4 ppm

Thank you, yes you're right, it is more important for the first OpAmp. Still, even then we're close to reference limited. I'd argue that I'd prefer reduced cost/current consumption to a factor of 2 in 1/f noise for this design.

Or, a different argument: it's actually quite hard to hit the reference's 0.25ppm noise floor, particularly in an environment with high thermal loads. I suspect that unless we keep the power dissipation down and maybe pot it, we'll measure a few times that. In which case, the OpAmps will contribute but probably not dominate.

dnadlinger, 2019-09-02:
> it's actually quite hard to hit the reference's 0.25ppm noise floor

Agreed; Fastino just flapping around in the (rack fan) will almost certainly be worse than 0.4 ppm; even more so the full system. I just wanted to point out that the design isn't optimised for low-frequency noise, in case someone thinks they'll be limited by it in their use case. May they speak now, or…

hartytp, 2019-09-03:
Tina TI simulation of the filter on stabilizer

[image]

With the OPAx156 TF is 
[image]

OPA197:
[image]

hartytp, 2019-09-03:
So, I'd actually argue that the filter performance isn't enough justification to go for the faster OpAmps. 

Anyone against going for an OPA2197 for the active filter (and also on stabilizer)?

hartytp, 2019-09-03:
I'm **much** more worried about power dissipation than a few dB on the active filter.

hartytp, 2019-09-03:
Slew rate is 20V/us, which is fine

dnadlinger, 2019-09-03:
We can also go to a multiple-feedback topology for better high-frequency attenuation, although I suppose we'd be more sensitive to part-to-part variations.

hartytp, 2019-09-03:
> We can also go to a multiple-feedback topology for better high-frequency attenuation, although I suppose we'd be more sensitive to part-to-part variations.

True. Feel free to suggest something ;)

jbqubit, 2019-09-03:
> Yeah, and looking at the strike-throughs on the Sinara team page,
> I'm not sure there are even any Yb-ers left.

There's quite a few Yb+ Sinara users. Currently using Sinara there are 5 traps at UMD and 3 traps at DOD  labs. Soon there will be 3 more Yb+ traps at UMD using Sinara. I'll pester people locally who are following the Sinara hardware discussions to add themselves to the sinara [wiki](https://github.com/sinara-hw/meta/wiki/Team#umd). 

>HOA 2 is rated to +/-20V according to the manual. 
>Might be worth checking if users of overly-bulky ions 
>like Yb+ actually make use of this capability? @jbqubit ?

Assuming 171Yb+, HOA-2.0, +/-1 V and potential optimization to maximum axial frequency the result is 700 kHz. +/-10 V would give 2.2 MHz, pretty much the bare minimum. I'd say +/- 20 V is desirable but not at the cost of missing other design objectives.  

> On the subject of asymmetric voltage ranges:

Agreed that this is an unwanted level of optimization. Let's keep this somewhat general purpose.

hartytp, 2019-09-03:
Let’s aim to do our best to support higher voltage ranges as a population option (above +-16V or so will need a different opamp). This will need different voltage set resistors for the regulators, different gain set and filter rcs is etc so it’s not something we’ll do now but interested parties can do it when the need arises

closed 2019-09-07 by hartytp