sinara-hw/Fastino #3: different ouput op amp

hartytp, 2019-08-30:
https://www.analog.com/media/en/technical-documentation/data-sheets/ADA4625-1-4625-2.pdf may also work and is half the price of the OpAmp on stabilizer. Slightly lower BW so need to re-do filter simulation.

@gkasprow do you have a spice model of the AFE of Stabilizer including all relevant noise sources?

gkasprow, 2019-08-30:
All simulations are [here](https://www.dropbox.com/sh/iv3qgrgfrsztj9w/AAB7AWFddpc1Nh-4yLUgz0Xma?dl=0) Filters were simulated with ADISimPE. The tool has noise analysis but I'm not sure if components have right noise models because the opamp model file is encrypted.

hartytp, 2019-08-30:
Thanks! I'll have a play with that.

dnadlinger, 2019-09-07:
See #4 and discussion in #2; OPA197 looks like a good candidate.

hartytp, 2019-09-10:
@gkasprow we've selected the OPA197 as our amplifier. Please close once that's on the schematic

gkasprow, 2019-09-10:
I used dual version, OPA2197. The schematic is already in releases - I replaced the old one.

closed 2019-09-10 by hartytp