sinara-hw/Fastino #5: adiabatic use case, f_DAC_clock > f_secular

jbqubit, 2019-09-03:
This morning I reviewed the [discussion](https://github.com/sinara-hw/Fastino/issues/1#) originating in 2018 where the idea of Fastino was first broached. As I understand the present consensus, Fastino is a many-channel DAC for adiabatic ion transport in many-electrode ion traps (eg HOA2). With the present DAC choice ad5542a (4 ch, 16-bit, 50 MHz SPI, no interpolation) we have something like f_DAC_clock = f_DAC_data = 50/4/16 = 780 kHz. 

The present consensus appears to neglect a rather important design criterion that was raised back in 2018, namely f_dac_clock > f_secular = 10 MHz. @hartytp [said](https://github.com/sinara-hw/Fastino/issues/1#issuecomment-415751385) 
> For the adiabatic use case, f_DAC only needs to be slightly above f_z.

The [2009 PRL by Blakestad et al](https://link.aps.org/doi/10.1103/PhysRevLett.102.153002) comes to the same conclusion [arXiv](https://arxiv.org/pdf/0901.0533.pdf). Recall that Blakestad was interested in adiabatic transport of 9Be+ in a 3D wafer trap. 

[image]

What happened to this consideration?

dtcallcock, 2019-09-03:
The ad5542a is single channel, so the f_DAC_clock = f_DAC_data = 50/16 = 3MSPS.

Clearly this will still not be good enough in some cases, which is why Shuttler is being developed.

hartytp, 2019-09-03:
> With the present DAC choice ad5542a (4 ch, 16-bit, 50 MHz SPI, no interpolation) we have something like f_DAC_clock = f_DAC_data = 50/4/16 = 780 kHz.

Nope, that's not the proposal. 32 channels, 2MSPS simultaneous updates on all channels.

> The present consensus appears to neglect a rather important design criterion that was raised back in 2018, namely f_dac_clock > f_secular = 10 MHz. @hartytp said

Firstly, there is the practical consideration that to get >=10MSPS you can't use a low-power, low-cost SPI DAC, but have to use a parallel/JESD monster. Then the pin-count/complexity means you end up with a design like Shuttler that's an order of magnitude more expensive, higher power dissipation, more complex, etc

Secondly, the main target here is axial frequencies bellow 2MHz and radial frequencies above 2MHz. So, even during shuttling/splitting there is no point where a mode hits the update rate. That clearly won't work for every use case, but it is what we're after.

closed 2019-09-03 by hartytp

jbqubit, 2019-09-03:
Ok. I misread the spec sheet; glad it's not a quad DAC. 

So, 50 MHz SPI implies 50/16 = 3.125 MSPS. So is it 2 MSPS or 3.125 MSPS? 

The noise power that annoyed Blakestad was at the DAC update frequency not the Nyquist. f_axial < 3.125 MHz seems like a fine compromise.

hartytp, 2019-09-03:
The hardware limit is 29.MSPS IIRC (you've got to take into account the entire SPI transaction including LDAC/CS etc). The proposed gateware implementation (see wiki) would run up to 2MSPS for 125MHz RTIO, but that could in principle be upgraded.

jbqubit, 2019-09-03:
You mean 2.9 MSPS, right?

hartytp, 2019-09-03:
yes