sinara-hw/Fastino #7: > 2.0 MSPS

jbqubit, 2019-09-03:
The proposed gateware for Fastino's AD5542A targets 2 MSPS. This is the most straightforward rate given a 125 MHz RTIO clock. @hartytp [recollects](https://github.com/sinara-hw/Fastino/issues/5#issuecomment-527612175) that the AD5542A supports up to 2.9 MSPS. As this raises the maximum possible secular frequency by nearly 50% it's worth exploring. 

Possible implementation... maintain 2 MSPS RTIO data rate with interpolation on local FPGA data rate closer to 2.9 MSPS. @jordens How involved would something like this be? Seems like it would increase the pipeline latency. Too bad it's not 3.0 MSPS.

hartytp, 2019-09-03:
@jbqubit are you thinking of funding this?

cjbe, 2019-09-03:
IMHO the active+passive filters should really kill the noise at f_sample, so although one would not want to run with secular frequencies bang on resonance with f_sample, this spur should be small / narrow enough that it is not problematic (i.e. run >1 kHz away, sweep over fast). @jbqubit I would be interested in your experience on this at NIST.

jordens, 2019-09-03:
There is vanishing power at f_sample in a sampled system. The Nyquist images potentially causing heating are guaranteed to be elsewhere.
This is a misconception of sampling.
Proper AA filtering and spectral shaping of the waveforms are the answer. Sample rates above twice the AA filter cutoff are inefficient and inadvisable, echoing what @cjbe says. Any good as filter will have sufficient attenuation at twice Nyquist to suppress the residual sample clock leakage. Waveform design can and needs to handle the rest by avoiding the sample clock.

@jbqubit let me know if and how you are interested in getting involved and I'd be happy to advise and suggest solutions.

jbqubit, 2019-09-04:
> @jbqubit are you thinking of funding this?

hartytp, 2019-09-04:
?

hartytp, 2019-09-06:
> There is vanishing power at f_sample in a sampled system. The Nyquist images potentially causing heating are guaranteed to be elsewhere.
This is a misconception of sampling.

For an ideal sampled system, yes, but there will be update glitches at f_sample which may or may not be problematic for a given application.

hartytp, 2019-09-06:
@jbqubit my aim here is that for the initial contract (see the Wiki) I want to go for a minimal implementation that gets us what we need in the lab, while making sure to keep it extendible e.g. to higher sample rates, or to things like the "optional" features on the Wiki.

2MSPs is fine for us and the increase in complexity/cost of going to 2.9MSPs isn't worth the hassle -- although if you or anyone else want to fund it as an extension, that's great. The limitation to sample rate is that if we want to run with the proposal on the Wiki (4 data lanes at f_rtio=125MHz) then we only get 1GBPs data rate, which gives 2MSPs. I can see at least a few different ways the sample rate could be upped:
- More data lanes. This would likely require use of the second EEM connector (which we will not use in the implementation). I'm currently keen to avoid this as far as possible since, until we have DDMA etc, EEMs on the master are more valuable than on the slaves, so high-density operation is preferable
- faster line rate: should be fine from a hw perspective, but slightly more complex gateware design
- interpolation on Fastino: assuming the Fastino FPGA can do this, that's another option, but it wouldn't really help in our case

jordens, 2019-09-06:
Yes. I mentioned sample clock leakage (you can also call that update glitches) above.
There are multiple possible approaches that could be looked at. From a fractional resampler over clock dithering or different approaches leading to higher data rates.

jordens, 2019-09-14:
not relevant for hardware. no reply to questions. closing.

closed 2019-09-14 by jordens