sinara-hw/Fastino #8: FPGA selection

gkasprow, 2019-09-03:
@jordens do you want to go for Artix 7 / SPartan 6 or ICE40 would be sufficient?
The biggest ICE40 is 8L LE, so if you are already close to this number with your gateware, It would be better to use the same chip as in Sayma RTM

jordens, 2019-09-04:
I don't think LUT/FF resources will be limiting. More worried about the timing and clocking. I am reasonably convinced the ice40hx8k should be fine but would like to prototype it on Kasli+Banker first.
Any interpolation or scaling would require multipliers but since Fastino is explicitly designed to stream the raw samples this can always be done upstream.

gkasprow, 2019-09-06:
I'm a bit worried about ICE40 working as a 125MHz serializer/deserializer. Before I start schematics, I'd like to know if it is feasible. In addition, IDELAY feature is often useful in such cases, especially with higher clock speeds.

gkasprow, 2019-09-07:
We can also go for Spartan 7 devices. They are nearly twice cheaper comparing with Artix 7. We don't need transceivers anyway. XC7S15-1FTGB196C costs 16 EUR and can be replaced by XC7S25 and XC7S50. It has 100 IO which should be enough.

gkasprow, 2019-09-07:
XC7A15T-1FTG256C  is 25 EUR, but offers 170IO and migration path up to XC7A100T.
The FPGA price would be negligible due to the amount of pricey analog stuff.
So IMHO we should choose Artix for this project to minimize risks and provide individual control of all DACs.

gkasprow, 2019-09-07:
We would have to go for XC7A15T-1CSG324C (31EUR) due to bank constraints in 256 package. I need one bank supplied from 2.5V which supports 16 LVDS lines.

gkasprow, 2019-09-07:
would there be any use

jordens, 2019-09-07:
I am pretty certain that an ice40hx8k will work. I'll prototype it on banker and test the fast serdes concept. It would also be very beneficial to use the ice40 as that gateware can be reused on banker and humpback.

gkasprow, 2019-09-07:
True. I already did schematics with Aritx chip but also have a version with ICE40.
I'm worrying only about one thing - with ICE 40 there is no migration path. Essentially, with ICE we can only do simple deserializer with buffering. We have only 250LE per channel. With Artix one can also do some DSP.

jordens, 2019-09-07:
Yes. But since we are explicitly designing this to always support full data rate over eem, the dsp can be pushed to the eem master. This also makes fastino agnostic of just that dsp use case.

gkasprow, 2019-09-07:
OK, So I'm switching to ICE40 version.

closed 2019-09-07 by gkasprow