sinara-hw/Fastino #10: reference distribution

hartytp, 2019-09-07:
- let's make sure we use star trace layouts and wide traces for reference distribution to minimize common impedance cross-talk
- how many DACs to we want to share a reference buffer? 4? 8? @gkasprow once the filter design is complete, can you simulate the cross talk (DAC and AC) due to the DACs + filters so we can see how bad this is

gkasprow, 2019-09-07:
@hartytp that's why I used 0R resistors to force such routing.
I did some simulation in the past and AFAIR two buffers would do the job.
I will play with it once again.

hartytp, 2019-09-07:
> @hartytp that's why I used 0R resistors to force such routing.

Aah, nice trick

> I did some simulation in the past and AFAIR two buffers would do the job.
> I will play with it once again.

When you're done can you upload the spice sim to git please?

NB now we must also take into account the reference subtraction in David's filters...

gkasprow, 2019-09-15:
I did a simulation with 8 loads running at 8 different frequencies from 10k to 1MHz.
[image]
The result over 20ms:
[image]
Rippless are below 20uVpp
[image]

gkasprow, 2019-09-16:
After replacing 10uF caps at the DAC reference inputs, with ceramic ones (10mOhms of ESR), the ripples are twice lower:
[image]
the worst-case event:
[image]

hartytp, 2019-09-16:
@gkasprow unlike Stabilizer, Fastino is mainly used open-loop. So, it's not just the AC effects we need to worry about (although, they are important) it's the DC ones too. The capacitors won't affect that.

What's the DC shift in output voltage on one channel as a function of the reference current drawn by the other channels?

gkasprow, 2019-09-16:
I do a DC sweep but I'm not able to measure any voltage drop on . Take into account that we have external opamp that trims the DC response.
The blue line is the LDO output, the green one is the DAC ref input. 
sweep 0-200uA
[image]

With increased current by 1000x
sweep 0-200mA
[image]
The circuit
[image]

hartytp, 2019-09-16:
Okay, great! So the green line is just the trace voltage drop? So, that would correspond to degraded INL (and maybe DNL as well if the current/DAC code curve is not smooth), but not to cross-talk?

gkasprow, 2019-09-16:
trace + 0R resistor.

gkasprow, 2019-09-16:
It's roughly 5uV within entire 0-200uA DAC current

hartytp, 2019-09-16:
nice!

How do you think the AC ripple compares with the DAC glitches after going through the filter? The glitches are huge, but the filter will smooth them out quite a bit.

[image]

gkasprow, 2019-09-16:
The reference glitches are nearly 3 orders of magnitude lower - 10uVpp vs 9mV

hartytp, 2019-09-16:
Right, but after they go through the filters, isn't it the glitch area that really matters?

gkasprow, 2019-09-16:
Yes, it is. It's hard to estimate the area above and below Vout, but they seem to be similar. So it should not affect the DC. Such DAC should be refreshed only when one updates its value to avoid such glitches.

hartytp, 2019-09-16:
@gkasprow one other question: if you replace the LT3042 with a 3045, does that make the glitches smaller? i.e. can we replace 2xLT3042 with 1x3045?

hartytp, 2019-09-16:
Other than that, this seems good. Let's go for 1 ref buffer per 8 channels.

gkasprow, 2019-09-16:
They have exactly the same data for load regulation, line regulation, dropout, etc. They differ mostly with max current and output noise.

hartytp, 2019-09-16:
hmm...I thought the 3045 had a lower output impedance, which would give lower glitches (more output pins IIRC)? But, maybe not...

gkasprow, 2019-09-16:
It has simply bigger transistor. And it is in current source configuration. So, for AC its output impedance is similar. For DC, it is kept low by feedback loop. Your statement would be true if the LDO had an emitter follower at its output.

hartytp, 2019-09-16:
Okay, thanks! Can you make sure to include the spice files for this in the repository somewhere. Then I think we can close. Obviously, just watch out for the trace routing from the ref buffer to the DACs (wide traces, star configuration, traces as short as possible etc etc).

gkasprow, 2019-09-16:
sure. This is how I route the references. I will make these traces even beefer
[image]

hartytp, 2019-09-20:
@gkasprow NB the reference current drawn by the subtractor stage varies between 0 (5V DAC output) and 5V/10k = 500uA (0V DAC output).

The max DAC current variations are relatively small [image] comparatively

So, we should be simulating for approx 0->500uA instead of the 0->200uA above.

Also, can you plot the step response (as well as the noise trace) for all channels switching from 0->500uA simultaneously. I know there is no DC error due to the OpAmp, but I'm worried about intermediate timescales where the reference impedance rises, but the capacitors are still not low impedance yet.

gkasprow, 2019-10-20:
True, it looks like this is slightly over-engineered.
I also made sure that the traces are thick enough. 
But such an approach should give us extremely low crosstalk.

hartytp, 2019-10-20:
@gkasprow So I know what to expect out of the final design, please could you post the following using your model:
(a) step response with all channels going from min to max ref current. I'd like to see this over a wide range of timescales, since the dynamics will be a bit complex; the impedance is low at high frequencies due to the capacitors, but rises with decreasing frequency as the capacitor impedance increases, then drops to very low values towards DC due to the OpAmp feedback. I'd like to check that the step response doesn't do anything nasty at intermediate timescales
(b) the same as before where you pulse all channels on/off at regular intervals

NB the reference current you previously used in your simulation was not correct, since it did not include the current drawn by the reference subtraction (which is now larger than the current drawn by the DAC itself). Please can you correct this before posting?

Thanks! I do not expect an issue with the current design, but worth checking the simulation one last time before finalising

gkasprow, 2019-10-27:
Here is the simulation with all 8 sources switching between 0 and 500uA with different frequencies.
One trace is LTC output, another is DAC input
[image]

0-500uA sweep
[image]

All DACs switching between 0 and 500uA with 1KHz frequency. One plot is LTC output, another is DAC input
[image]

jordens, 2019-10-27:
20uV max. Excellent.

hartytp, 2019-10-27:
@gkasprow wow, that's better than I'd hoped! We probably should have cut the number of reference buffers to save some pennies ;)

Thanks so much for doing the detailed simulations, that will make me sleep much better :) 20uV pk-pk on 20V is 1ppm, which is <0.1LSB pk-pk, so pretty impressive!

One final question: if you let the final simulation run longer (say a few ms), is the change monotonic? Or, is there some time scale where the capacitors start do discharge but the DC feedback is not quite working yet?

closed 2019-10-27 by hartytp

gkasprow, 2019-10-27:
I simulated it over 1s or so and started plotting in steady-state. I also limited the LTC filtering capacitance to make it converge faster. I will do such a simulation for curiosity.

gkasprow, 2019-10-27:
edit. The simulation was 30ms and plotting started after 10ms. Once I made it running for 1s.

gkasprow, 2019-10-27:
It took a lot of time...
[image]

hartytp, 2019-10-27:
Nice! Nothing funny going on at any timescale. Good!