sinara-hw/Fastino #9: schematic review

hartytp, 2019-09-07:
Wow! That was quick Greg!

hartytp, 2019-09-07:
@gkasprow I won't do a careful design review until we've finished the high-level design, so am not checking supplies etc yet

I'm also not looking at all at the digital section, since @jordens is doing that.

So, really all I'm looking at right now is the reference, DACs and filters. Apart from the currently open issues, that looks good to me!

gkasprow, 2019-09-07:
It looks like before we proceed with layout, @jordens need to make sure that ICE40 will do its job.
Right now, I have 2 versions of schematics - with Artix and ICE40.

jordens, 2019-09-24:
Ice40 will work. Digital aspects are in #16.

hartytp, 2019-09-26:
@gkasprow what's the timeline for getting a complete schematic RC ready?

gkasprow, 2019-09-27:
2 weeks

hartytp, 2019-09-27:
Cool, thanks

hartytp, 2019-10-15:
@gkasprow what's the state of this? We really need to get these boards to manufacture in the next week or so if possible.

gkasprow, 2019-10-15:
I'm fighting with the layout. I will finish it till the end of the wee,

jordens, 2019-10-16:
If this is too dense, let us know. We can relax the design.

gkasprow, 2019-10-20:
The layout is done. I will publish it for review in a moment

hartytp, 2019-10-20:
Cool! Let me know when you have a new rc PDF release

gkasprow, 2019-10-20:
still needs to do SI/PI simulations
PDF is here https://github.com/sinara-hw/Fastino/releases/tag/v1.0rc2

jordens, 2019-10-21:
Apart from CFG_RELOAD, I2C and HWREV, what changed on the FPGA pinout since the last pinout you posted?

gkasprow, 2019-10-21:
I think I shifted some of the DAC lines in the IOB to make pins for CLK

jordens, 2019-10-21:
There is a RED LED for that already.

closed 2019-10-24 by hartytp