sinara-hw/Fastino #14: SMPS/LDO power budget

hartytp, 2019-09-16:
To do: check the power budget carefully and ensure we have enough margin. The +-21V converter we currently use is only 6W and it would be easy to exceed that will all channels driving a capacitive load due to un-terminated cabling.

gkasprow, 2019-09-16:
Yes, I plan to replace it with 10W version which is slightly bigger. Do we need much more power than that?

hartytp, 2019-09-16:
I don't think so, but let's double check the power budget.

FWIW, if we went for higher bias current OpAmps, this could be a real problem. But, IMHO EEMs really shouldn't be dissipating much more than 10W or the thermal management will become a killer!

We should consider adding a thermal switch somewhere however.

gkasprow, 2019-09-16:
the thermal switch (IC7) is already on the schematics:)

gkasprow, 2019-09-16:
With the existing design, we can safely draw +/- 6mA from all channel at once. I assume that we operate with AC signal that consumes 6mA from both +12V and -12V rails. The entire board power consumption is 13W in such case. Without the load, it consumes 7W.

gkasprow, 2019-09-16:
Edit: 12.7W with load,7.5W without the load.

hartytp, 2019-09-16:
R=V/I=10V/6mA=1.7k

So the min DC load is approx 2k. That's fine for us (we will drive basically only the cable capacitance) but worth noting the current limit on the schematic.

I guess if the user shorts too many channels the PSU will current limit and undefined behavior will ensue.

gkasprow, 2019-09-16:
The LDO is 250mA, so it should trip first. It has PG output so we could try to make use of it...
In case we need more power, [15W PSU](https://www.cincon.com/productdownload/EC3SB-V18.pdf) is available with similar dimensions. Its output can be trimmed to +/-13V. It costs similar to CC10-1212DF-E

gkasprow, 2019-09-16:
edit, LDOs are 200mA

hartytp, 2019-09-16:
Cool. Well, let's do the power budget and then make sure we have some margin.

I think that if users want high speed amps for filters etc, or wider voltage ranges, they should make a small daughter card to do that. Otherwise the power consumption on Fastino is just too high

gkasprow, 2019-09-16:
The detailed power budget is [here](https://www.dropbox.com/s/5416p1zyza7l3iu/Fastino_Power_Budget.xlsx?dl=0)

hartytp, 2019-09-16:
thanks.

Well, pick whichever LDOs you think are most suitable and then I'll double check when I do a design review.

jordens, 2019-09-24:
* The LTC6655 only has 0.2V dropout when operated at the 60µA we are looking at. It could be powered from P5V5A or even P12V0A and the P7V0A rail could be dropped.

* Do you really need to burn `4*18mA*5V~400mW` idle power from the reference buffers?

gkasprow, 2019-10-20:
true, 400mW is a lot. I increased these resistors to 1k.
I also removed 7V rail. Good catch.

gkasprow, 2019-10-20:
I also used higher power 10W DC/DC and increased LDO currents to 400mA to give some extra margin to the AFE.
I used LT3045 and LT3094. @hartytp please check IC4 and IC3.

hartytp, 2019-10-20:
Cool, thanks for fixing! Reopening this to remind myself to review the changes. Will close again after review

hartytp, 2019-10-24:
NB agreed that the 10W SMPS is probably overkill, but the lower power one could be a bit marginal with heavy output switching so seems worth keeping the bigger one to give us more margin...

closed 2019-10-24 by hartytp