sinara-hw/Fastino #15: Link design

jordens, 2019-09-20:
I have a prototype for the high speed data link over EEM at
https://github.com/quartiq/fastino
combined with the Kasli prototype at https://github.com/m-labs/artiq/tree/fastino

This works well (<5e-13 BER so far) without an ultimate level of tuning on a typically EEM ribbon cable length of 0.5m and the boards strewn out across the table.
With 3.5m ribbon there is about 1e-9 BER. Still, there are a couple of knobs to tweak that should improve this even for long ribbon cables.

The projected DAC sample rate is 2.55 MS/s each channel. I would leave the filter where it is as this improves the signal quality and puts clock leackage close to the "notch".

Also with this design (due to the SPI-data-word clock relation) there is currently a 14 ns pk-pk triangular deterministic jitter on the DAC sample timing at f_sample/3. This is irrelevant for most applications but it would become visible as an aliased spur on periodic signals.
If this becomes problematic, there are ways around it.

This supports the case for the ice40. Several change requests to the pinout to follow.

gkasprow, 2019-09-20:
IDC ribbon cable has 120Ohm impedance of single wire in G-S-G configuration. We use it in G-S-S-G config which gives value closer to 100Ohm. DIfferent ribbon cables have different impedances. There are ones with 105Ohm which should be avoided.

hartytp, 2019-09-20:
Very nice. Do we get a red LED or something on checksum errors?

jordens, 2019-09-20:
Interesting. My go-to ribbon cable has 102 ohm in gsg config. http://multimedia.3m.com/mws/media/22052O/3mtm-round-conductor-flat-cable-050in-3365-series-ts0080.pdf

gkasprow, 2019-09-20:
And this cable gives 170Ohm in a balanced configuration. I was wrong in the post above.
We need SGS impedance as low as possible, probably 85Ohm which would be closer to 100Ohm of differential impedance.
People are using  25cm .05 ribbon for PCIe so we should not observe issues at 250MHz.

gkasprow, 2019-09-20:
for high speed one can use this [kind of cable](http://multimedia.3m.com/mws/media/22097O/3mtm-shielded-jktd-flat-cable-050-3517-series-ts0069.pdf)

jordens, 2019-09-20:
You can get a led. Counters for bitslip and crc errors that can be cleared are already there.

hartytp, 2019-09-20:
:)

closed 2019-09-24 by jordens