sinara-hw/Fastino #16: FPGA pinout

jordens, 2019-09-23:
* The B side of a LVDS input is the positive one (https://github.com/sinara-hw/Humpback/issues/9)
* Connect the EEMR0_0 to the GBIN7/IOL_13 pair and EEMR1_0 to the GBIN6/IOL_14 pair
* We are not completely following the ice40 recommendation for LVDS output resistor network. They recommend 150/140 while we have 140/100 (the LVDS input recommendation) which gives less differential swing. Also the IDC impedance is not 100 but higher. Maybe 150/140 is actually better in our case, even for LVDS inputs? Or is that too risky? @gkasprow 
* A couple limitations on pin swapping of the DAC SPI buses
  * Make sure each DAC covers exactly 2 PIO pairs with its four signals. Which pins of the R/T/B side are part of which PIO pair seems a bit hard to find, the only location I can see right now is the [icebox source](https://github.com/cliffordwolf/icestorm/blob/0ec00d892a91cc68e45479b46161f649caea2933/icebox/icebox.py#L4721-L4928). You can also tell from the missing IOx_yyy numbers in the odd/even pairs. The unpaired pins (the other PIO side is not bonded out) are: E14 M15 T13 M7 B4 J12 R4 J11 D14 B8 K15 D8 R16 A7 T15 R6 C13 A9. Use those for I2C, switches, hwrev, LEDS.
  * Make sure that the CS and LDAC signals share a PIO pair and that SDx and CLK share the other.
  * Make sure the DACs split up evenly across the banks. I.e. 12 per bank max, otherwise the limited clock enable routing might be annoying later. And it distributes the simultaneous switching noise and ground bounce a bit.
  * Try to keep each group of 4 DAC channels within one bank and not split them.
  * Use M9 R9 B9 F7 as outputs only (LEDs, DAC_SPI_CLRn). They otherwise block the PLLs.
* Also use the left bank (2.5V) for LEDs, switches, hw_rev.
* Please route the SPI CLK lines away from the data/cs/ldac lines if possible.
* Due to PLL/GB routing, connect CLK25 to GBIN2 or GBIN3.
* Check with @hartytp as to the color of the LEDs, might be worth throwing a couple red ones in there.
* The I2C pullups on the FPGA side (2k4) are too strong IMO. We don't expect Fastino to be I2C master to downstream EEMs. Also connect the FPGA to Both I2C buses (but separately), not just one.
* I2C1 needs pullups on the EEM/eeprom side (10k).
* No need for the CFG_RELOAD pin and resistor. The SB_WARMBOOT primitive can do that internally.
* Add 2 hw_rev resistors.
* We are planning to spend a significant time update all 32 DACs at full rate (50 MHz SPI, 32 channels, 2.55 MS/s each channel). Do you think this will be problematic in terms of digital feedthrough, ground bounce, simultaneous switching, filter crosstalk?

gkasprow, 2019-10-16:
I did such termination on purpose.
The datasheet recommends such values for output only.
When we switch it to input mode, it is terminated with 140Ohm instead of 100R.
That's why I used 100R and lowered the other resistors.
Anyway, with 2.5V we still have 650mV pp. We can make it better by changing 100R to 120R which fits better the cable impedance.

closed 2019-10-16 by gkasprow

jordens, 2019-10-17:
Yes. I know why you did that. But it leads to a low output impedance. And the datasheet does document the recommended input impedance/network. 120 Rp is fine with me. Still seems a bit low compared to the IDC cable.