sinara-hw/Fastino #19: FPGA clocking

jordens, 2019-09-24:
* Connect the EEMR0_0 to the GBIN7/IOL_13 pair and EEMR1_0 to the GBIN6/IOL_14 pair
* Due to PLL/GB routing, connect CLK25 to GBIN2 or GBIN3.
* Use M9 R9 B9 F7 as outputs only (LEDs, DAC_SPI_CLRn). They otherwise block the PLLs.

gkasprow, 2019-10-06:
@jordens do you need CBSEL lines? I have to use them if we want to connect the clock to GBIN2 or 3
Cannot we use GBIN4 or 5 ?

gkasprow, 2019-10-06:
@jordens is 25MHz clock fine? Maybe other frequency would be better?

jordens, 2019-10-06:
There is a pll in the gbin4/5 tile that's blocked if these are used as inputs (unless the inputs go to the pll).
Clock frequency doesn't matter. You can choose whatever is convenient. This will be clocked from eem.

jordens, 2019-10-06:
I'll check cbsel

jordens, 2019-10-07:
Correction. GBIN5 (IOB_81) and IOB_80 can not be used as inputs if the PLL is used. Same for GBIN0 (IOT_198) and IOT_199.
So you can connect the 25 MHz clock to any of GBIN4/3/2/1.
(Same as https://github.com/sinara-hw/Banker/issues/7)

The CBSEL lines might be nice for using different bitstreams. Keep them if you can.

gkasprow, 2019-10-07:
@jordens With GBIN4 as the clock input, I don't need CBSEL lines and can leave them for bank select feature.
This is a proposed pin assignment
[image]
[image]

And here is the pin table that I merged with PIO locations
[iCE40PinoutHX8K.xlsx](https://github.com/sinara-hw/Fastino/files/3696566/iCE40PinoutHX8K.xlsx)

jordens, 2019-10-07:
But you used GBIN5 for the clock.

jordens, 2019-10-07:
* The mixing of hypothetical channel groups among the banks is probably OK. Having a more symmetric assignment would allow a more symmetric choice of the half cycle switching shifts (re #22 ). But this is OK from my side since that symmetry it's not a requirement.
* Splitting channel 13 across banks 0 and 1 is a bit annoying. But also probably OK.
* The banks have 11.5, 11.5, 9 DAC channels each. This is fine with me.

gkasprow, 2019-10-07:
I'm not sure if I can route it without splitting the channel 13

gkasprow, 2019-10-07:
True, I switched the clock to GBIN4. 
@jordens here is an updated pin assignment. Can you make sure that all compiles so I would finish routing?
[Uploading Fastino_FPGA_signals.xlsx…]()

jordens, 2019-10-07:
your upload didn't work.

gkasprow, 2019-10-07:
it should work now.

jordens, 2019-10-07:
works fine with the prototype gateware.

gkasprow, 2019-10-07:
Cool!

closed 2019-10-07 by gkasprow