sinara-hw/Fastino #24: v1.0rc2 schematic review

hartytp, 2019-10-24:
@jordens have you completed your review of the schematics? I'll do mine now.

hartytp, 2019-10-24:
- [ ] Cosmetic: why do R170 and R171 have the value "undefined" rather than just being DNPd in this variant with a red cross? Doesn't that screw the bom/make populating them later on harder (since the value they should have in a populated variant is not shown)
- Maybe I'm being blind, but I don't see any red user LEDs.
- Cosmetic: why is the power budget an excel file instead of the pretty looking annotations we normally have? Doesn't really matter, I suppose, but nice to keep all the designs consistent and pretty!
- [ ] +-12V regulator set resistors look an order of magnitude too high to me. 

[image]

hartytp, 2019-10-24:
- [ ] The LT3045 and LT3094 require a minimum of 10uF input capacitance and recommend low ESR capacitors. Are you sure the 1uF ceramic + 10uF electrolytic will do, or should we replace the 1uF with 10uF ceramic?

hartytp, 2019-10-24:
- [ ] for the reference buffers, the set capacitors should be ceramic as recommended by LTC

hartytp, 2019-10-24:
- [ ] I believe you got the `Ilim` wrong for the LT3094

[image]

jordens, 2019-10-24:
I'm done with my part schematics review.

If you search the PDF for "RED" you'll get plenty of meaningful hits. LD1D and LD1C are red.

The power budget obviously needs to be taken with a grain of salt. (a) An opamp doesn't simultaneously load both rails with maximum current. (b) A large part of that power would be dissipated elsewhere (c) All channels at maximum current is not within the target use case.
Baseline power consumption is probably more like 4-5 W.

hartytp, 2019-10-24:
> If you search the PDF for "RED" you'll get plenty of meaningful hits. LD1D and LD1C are red.

Aah, thanks, I'd mised LED_RED.

> The power budget obviously needs to be taken with a grain of salt. (a) An opamp doesn't simultaneously load both rails with maximum current. (b) A large part of that power would be dissipated elsewhere (c) All channels at maximum current is not within the target use case.
Baseline power consumption is probably more like 4-5 W.

Yes, these power budgets always end up being very pessimistic. Still, they're useful to have...

hartytp, 2019-10-24:
>@gkasprow not that it really matters, but why are we not using the data sheet recommended values for the passives on IC18?

Same for IC90

hartytp, 2019-10-24:
@gkasprow if you still have problems with the layout, I'm fine scrapping the CMCs.

hartytp, 2019-10-24:
@dnadlinger are you happy with the current filter implementation? Didn't we agree to DNP C3?

[image]

hartytp, 2019-10-24:
- [ ] these annotations are incorrect (x4 gain and +-2*Vref)

[image]

hartytp, 2019-10-24:
- [ ] Is this compensation capacitor optimized for the current OpAmp choice? 
[image]

hartytp, 2019-10-24:
- [ ] Do we need to increase P5V0A to 5V1 or something to ensure that the DAC Vdd is always higher than Vref (even when taking component tolerances into account)? Before doing this, please confirm that there is nothing else on that rail that can't handle the change in voltage.

dnadlinger, 2019-10-24:
@hartytp: The issues are you pointed out are valid. As for C3, it does (in SPICE-land) buy you 16 dB extra attenuation at 4 MHz, so I'm 50:50 on whether to populate it initially to see how bad drifts and channel-channel spread really is vs. doing so by hand on a single board.

hartytp, 2019-10-24:
hmm...or can we just scrap P5V0A and use only P5V5A instead?

hartytp, 2019-10-24:
(maybe not if there are sequencing issues)?

hartytp, 2019-10-24:
> As for C3, it does (in SPICE-land) buy you 16 dB extra attenuation at 4 MHz, so I'm 50:50 on whether to populate it initially to see how bad drifts and channel-channel spread really are vs. doing so by hand on a single board.

Okay, let's keep it for the initial version.

hartytp, 2019-10-24:
[image]
 what P1V0?

hartytp, 2019-10-24:
what is P3V3A? I don't see a source for it? Or, am I missing something?

hartytp, 2019-10-24:
Should IC12 EN be tied to P5V5A PG?

hartytp, 2019-10-24:
[image]

We don't need D22A since there is no way of powering Fastino without the EEM connector

hartytp, 2019-10-24:
Shouldn't L6 be the same as Stabilizer?

hartytp, 2019-10-24:
[image]
 Sequencing isn't correct (mentions rails that don't exist)

hartytp, 2019-10-24:
Okay @gkasprow I've finished my review. Looks good, only some minor fixes.

Any chance we can send this to manufacture on Monday?

gkasprow, 2019-10-25:
I will do my best to send it on Monday. The same with Stabilizer.

hartytp, 2019-10-25:
Amazing! Thanks!

hartytp, 2019-10-25:
The design looks good by the way! I'm very happy with the way it's turned out