sinara-hw/Fastino #26: Rethink power sequencing

hartytp, 2019-10-24:
The power sequencing was copied (I assume) from Stabilizer and is way overcomplicated for our need. Not least because all our LDOs now have PG/EN lines!

My comments are:
- [ ] I think (please double check this) we can scrap P5V0A and run the DACs of P5V5A. That simplifies things
- [ ] I don't believe any particular sequencing is needed now. 

@jordens am I correct in thinking that the ICE40 pins are all inputs/high-z while CRESET_B is asserted, meaning the FPGA can't drive the DAC outputs before the DAC supply is valid?

hartytp, 2019-10-24:
- [ ] hmmm...do we need to keep the FPGA reset asserted until the P1V2 and P2V5 lines are stable?

jordens, 2019-10-24:
Apart from a few special purpose pins (CDONE etc) they are all high-z.

hartytp, 2019-10-24:
@jordens do you see any other potential sequencing issues I've missed? AFAICT, the above points should cover everything and simplify the design

hartytp, 2019-10-24:
LT3045: max EN 22V, threshold is 1.24V
LT3094: max is 22V to gnd or 30V to Vin, threshold is 1.26V
TPS62175: max EN is vin + 0.3 (so 12V), threshold is 0.9V
NCP3170ADR2G: max EN is Vin + 0.3, threshold is 1.4V
AUR9705: max EN is Vin + 0.3, threshold is 1.5V
LTC6655: max EN is Vin + 0.3V, threshold is 2V

LT3045: max PG is 22V
LT3094: max PG is 19V (30V-13V) 
TPS62175: PG max is 7V
NCP3170ADR2G: max PG is Vin + 0.3V

So, it looks like we can tie all the PG lines together with a weak pull up to 3V3 to limit current transients (or if we want to be paranoid, stick diodes between PG and Vin lines to stop PG going above Vin, but a 100k pull-up seems easier).

jordens, 2019-10-24:
All good from my side. If the digital part (FPGA) is carried from Banker/Humpback. No special sequencing.

gkasprow, 2019-10-27:
Right, we don't have any ADCs that could be damaged by wrong sequencing.

gkasprow, 2019-10-27:
One sequencing we need is to make sure DACs wakeup first to not cause the latchup from FPGA data lines. But we already have it.
We may also want to make sure that 12V wake up after the DACs outputs are stable to limit huge +/-12V transients at the outputs. So, I'd leave the power sequencing as it is.

hartytp, 2019-10-27:
I think the simpler sequencing (see my comment above)would be much cleaner and easier to understand and debug.

It ensures the fpga is held in reset until the dac supply is stable. It ramps the reference up slowly to avoid output transients

gkasprow, 2019-10-27:
I'm not sure that the reference will ramp smoothly. It would be true without DC stabilization loop

gkasprow, 2019-10-27:
I would even turn on the AFE supply from FPGA after we make sure that DACs have mid-range output state. Anyway, I think we can scrap the LM393 circuit.

hartytp, 2019-10-27:
Well, even so, output transients don’t worry me so much. Many opamps have transients even with their inputs grounded. I am happy assuming the load can handle +- 12V

hartytp, 2019-10-27:
Okay, yes I’d be happy controlling the afe supply from the fpga. Anyway the dac output is high z so won’t damage the opamp input in case of sequencing errors.

gkasprow, 2019-10-27:
OK, let's go for it. the simpler, the better. I remember a discussion about DIO and did my best to ensure there are no transients at the outputs during power-up.

gkasprow, 2019-10-27:
I'm not worrying about the DACs but about the load:)

gkasprow, 2019-10-27:
So, now we have the following sequencing
1. P5V4 wakes up first, powers DACs
2. then FPGA gets power to avoid DAC latchup
3.  Then DAC reference is enabled and FPGA is configured.
4. Once FPGA resets DACs, +/2 12V AFE is enabled.

gkasprow, 2019-10-27:
In the end, the front panel PG LED is on using +/-12V LDOs PG.

closed 2019-10-27 by gkasprow