sinara-hw/Fastino #30: LDO input capacitance

hartytp, 2019-10-24:
The LT3045 and LT3094 require a minimum of 10uF input capacitance and recommend low ESR capacitors. Are you sure the 1uF ceramic + 10uF electrolytic will do, or should we replace the 1uF with 10uF ceramic?

Also applies to ref buffers

hartytp, 2019-10-24:
NB I know we've had this discussion before, but the LT3045 set capacitor is recommended to be ceramic. I guess we don't care in this case as we're not relying on this to be massively high stability, but worth noting 
[image]

hartytp, 2019-10-24:
- [ ]  for the reference buffers, the set capacitors should be ceramic as recommended by LTC

gkasprow, 2019-10-27:
what about the noise? Ceramic ones like X7R are good microphones...

gkasprow, 2019-10-27:
I changed the input caps to 22uF

gkasprow, 2019-10-27:
@hartytp what about if we solder ceramic capacitor in one channel and compare results between the channels to see if such solution pickups the noise?

hartytp, 2019-10-27:
> what about the noise? Ceramic ones like X7R are good microphones...

Here's the way I look at it...we have two potential imperfections -- microphonics or leakage current -- and we have to pick one. Which one is worse depends completely on the application.

For the kind of lab environment we use Sinara in (e.g. mounted in a 19'' rack with a fan) I've never observed issues due to microphonics (we have many designs with LT6655/LT304x with large ceramic capacitors which are ultra-low noise). So, I don't find this to be a problem in our case. I can imagine that equipment going into a van or something would be very different.

On the other hand, temperature-dependent leakage can be a real problem where DC stability is required. Do the capacitors you use have a specification for the leakage temp co? I imagine this will dominate the stability of the circuit.

Having said that, in this case the leakage is almost certainly less of an issue because it is stabilised by DC feedback for the critical references. Microphonics can be more of an issue as they have more high-frequency content.

Note that we have a large ceramic on the LT6655 output, so we are using both in this design!

Anyway, looking again, i'm happy with the current situation, so let's leave it as is.

closed 2019-10-27 by hartytp