sinara-hw/Fastino #39: design review

hartytp, 2019-10-27:
- [ ] power budget needs updating to reflect the fact that we now have a single 5v4 rail, not 5V0 and 5V5
- [ ] why have you annotated the `_LX` nets? what do you connect to them?
- [ ] since you're now using the PG output of IC13, I assume you meant to add a PGFB divider there as well?
- [ ] Don't the ref buffers need 10uF ceramic input caps to ensure stability? Currently they are electrolytic, which could be too high ESR/ESL 
- [ ] ref buffer Ilim still does not match the annotation (data sheet says it's 125mA/kOhm, but it's a 220Ohm resistor, which does not match the 27.5mA specified current limit)

Other than that, all looks good to me! Assuming @jordens has no final feedback, let's send.

@jordens NB the new requirement to enable the AFE power supply from the FPGA.

hartytp, 2019-10-27:
@gkasprow one other thing: the 4 special LEDs on the FP have two-letter names, which can be a bit unclear to new users. Could we have something like an annotation of the first page that explains the functions of these LEDs?

gkasprow, 2019-10-27:
[image]
PG - power good
FD - FPGA done
OT - overtermp
UL - User LED

I placed it also on the schematic

gkasprow, 2019-10-27:
_LX nets are for Hyperlynx to know where the power lines are. I can remove after simulations.

closed 2019-10-27 by gkasprow

hartytp, 2019-10-27:
If you haven’t finalised the files can we swap ul for err or something. It will be a crc error on the data link

hartytp, 2019-10-27:
Anyway thanks again for another really lovely design! Very much looking forward to testing it on ions

gkasprow, 2019-10-27:
OK, ERR is fine.

commit f0c3314b (2019-10-27): v1.1rc3 fixed #39