sinara-hw/Fastino #42: Fastino testing

gkasprow, 2019-12-10:
I wrote a simple VHDL code that blinks front panel LED, turns the AFE power on and loads 32-nd DAC with some data. All of these work fine. All voltages have correct values so I think I can send the boards for further testing

gkasprow, 2019-12-10:
That's not that easy. There is a problem with output stage gain. When the DAC gives 2.5V, w get 5V at the output. When it gives 5V, there is 11.9 at the output

gkasprow, 2019-12-10:
the gain is OK, but there is a problem with output stage bias voltage. It is not 2.5V.

hartytp, 2019-12-10:
oops...what did we mess up?

hartytp, 2019-12-11:
Did the test points show 5V for the reference voltage?

hartytp, 2019-12-11:
> That's not that easy. There is a problem with output stage gain. When the DAC gives 2.5V, w get 5V at the output. When it gives 5V, there is 11.9 at the output

Doing the maths in my head, that exactly matches what I'd expect based on the resistor values we have populated. So, it looks like we messed up the new output filter :(

Will have to think about what the right fix for this is.

hartytp, 2019-12-11:
Original filter design is here https://github.com/sinara-hw/Fastino/issues/4#issuecomment-529139058

[image]

hartytp, 2019-12-11:
Here is what we have in Fastino:

[image]

So it appears to be a copy-paste error. The Vref resistor should be the low value one, not the resistor to ground.

hartytp, 2019-12-11:
Looking over the filter design issue, it seems we decided to change the resistor values to 10k for a good reason https://github.com/sinara-hw/Fastino/issues/4#issuecomment-529142111

i haven't double checked the compensation capacitor value yet. @gkasprow if you have time can you fix the resistive divider one one channel and take a step response? That will let us check if the compensation is about right.

hartytp, 2019-12-11:
Anyway...this looks like a simple uncaught copy-paste error in the schematic. What do we do? Is it feasible to get this patched on all channels that have been produced by TS?

gkasprow, 2019-12-11:
we need to swap 2 resistors in the SOIC package.
We have one free 5V resistor that helps a bit. With a piece of thin wire and a lot of manual work we can patch it.

gkasprow, 2019-12-17:
The easiest way to go is to use an unused resistor between pads 2 and 7.
The operations to perform:
1. lift leg 1 of RN5
2. cut or lift leg 6 of RN5
3. short leg 7 with leg 8 of RN5
4. short leg2(lad2) with pad1 (but not lifted leg2)
5. with short piece of wire short leg 1 with pad 6.
6. repeat steps 1-5 32 times 

I modified two amps and they generate correct 0V at the output with 2.5V at the DAC input.+

[image]

dnadlinger, 2019-12-17:
Oh dear, that sounds like a lot of work for 4 * 32 (?) channels! (I did point this out here https://github.com/sinara-hw/Fastino/issues/4#issuecomment-529154162, but unfortunately didn't look over the final schematics… :/)

gkasprow, 2019-12-17:
Unmodified boards are still working, only output is shifted by 2.5V offset

hartytp, 2019-12-18:
Okay good! So that’s basic testing complete. Good work!

 Now the priority is to get @jordens a patched board ASAP so he can complete software development. I’ll coordinate getting our boards patched with ts.

closed 2019-12-18 by hartytp

gkasprow, 2019-12-22:
I think I forgot to publish the photo
[image]