sinara-hw/Fastino #48: i2c with FPGA unconfigured

jordens, 2020-01-09:
* if the FPGA is not configured it does not enable the AFE power
* if the AFE power is not enabled, the parasitic diodes in the mosfets T6-T9 pull down the I2C busses
* this makes configuration via I2C impossible

jordens, 2020-01-09:
this may also apply to other boards where I2C is hooked up via a target-controlled switch to a configurable target.

jordens, 2020-01-09:
In fact i don't know why the source side is pulled down. The FPGA pins should be high-z

pathfinder49, 2020-01-24:
@gkasprow As far as I can tell, P12V0A and N12V0A are off until the FPGA is configured. This disconnects the i2c, preventing configuration.

[image]

gkasprow, 2020-01-24:
True, the idea was to disconnect I2C from FPGA until it is configured properly to not block the bus.

pathfinder49, 2020-01-24:
To avoid bypassing the power sequencing, I tried de-soldering R48 and connecting to REF_SHDN instead. This allows the 12V supply to start. However, En_AFE_PWR appears to be pulled low during flashing. Consequently the I2C disconnects again when flashing. Have you got another suggestion?

jordens, 2020-01-24:
No no. That's not the problem. The FPGA is not involved with I2C during flashing. Having it disconnected is completely fine. It's in fact the intention.

jordens, 2020-01-24:
There is also a faint glow on the P12V0A LED (and I think even N12V0A) with EN_AFE_PWR low which should not be there AFAIR.

jordens, 2020-01-24:
Additionally something is weird (compared to e.g. Banker) with the FPGA not loading its configuration (no CDONE) when reset (CRESET) via the I2C-SPI/GPIO extender. May be the same, may be something different.

pathfinder49, 2020-01-24:
> There is also a faint glow on the P12V0A LED (and I think even N12V0A) with EN_AFE_PWR low which should not be there AFAIR.

When EN_AFE_PWR is pulled to 0 V I measure no voltage on either P12V0A or N12V0A.
However, when unmodified EN_AFE_PWR is at 0.56 V. I then measure +2 V on the P12V0A and +0.6 V on N12V0A whith the P12V0A LED glowing slightly.

gkasprow, 2020-01-24:
I don't have any PCB to look at this.
Remove R42 and FPGA loading will work. At the moment the FPGA is loaded by PG signal

gkasprow, 2020-01-24:
the EN_AFE_PWR has 10k pulldown. Don't you have some FPGA pullup? It looks like we have to decrease the R48 to let's say 2k2

pathfinder49, 2020-01-24:
> No no. That's not the problem. The FPGA is not involved with I2C during flashing. Having it disconnected is completely fine. It's in fact the intention.

Excuse my ignorance, I thought the FPGA was flashed via I2C here. Am I missing something?
[image]

jordens, 2020-01-24:
The fpga has weak pullups when unconfigured. May need lower pulldown.
Still doesn't explain the failure to configure by creset though.

jordens, 2020-01-24:
The flash is flashed by i2c. The fpga is held in reset and is out of the loop.

pathfinder49, 2020-01-24:
The EEM0_I2C signal is switched off before this point as far as I can tell.

gkasprow, 2020-01-24:
The EEM I2C is routed to EEPROMs only. After FPGA wakes up, it enables P12V0A and enables access to the rest of the I2C tree.
This also means that the I2C extender is not connected to the EEM I2C until FPGA gets configured. THis needs to be fixed, the I2C extender should always be connected to the EEM I2C

gkasprow, 2020-01-24:
This also means that the I2C extender must be suplied from P3V3MP to not block the bus.

gkasprow, 2020-01-24:
as a temporary fix, short T6 and T9 (drain with source, pin 2 with pin 3)

jordens, 2020-01-24:
No. That's also irrelevant.

gkasprow, 2020-01-24:
@jordens do you mean that with unconfigured FPGA you can't access EEPROM?
With P12V0A high enough to make the LEDs glowing, it is enough to enable the MOSFET I2C switches which connect the bus to the FPGA. If the FPGA helds the IO pins low for some reason, the bus will be blocked.

jordens, 2020-01-24:
Yes. As the top post describes.

The fpga is not needed for programming. And the fpga does not drive the i2c bus. Not when configured and not when reset. Still the i2c bus is broken. And it's broken when afe pwr is **disabled**

jordens, 2020-01-24:
For me this also affected the lm75, not only the spi converter.

gkasprow, 2020-01-24:
LM75 is connected directly to the EEM connector. I'd removeT6, T9 and check if that helps.

pathfinder49, 2020-03-17:
@hartytp @gkasprow I'm just finishing the PR for the new Fastino revision. I've addressed most known issues. However I'm not sure what the way forward is here.

It would be good to arrive at a conclusion such that this can be fixed.

hartytp, 2020-03-17:
@pathfinder49 I don't feel qualified to make this call. The decision should come from @gkasprow or @jordens

pathfinder49, 2020-03-19:
@gkasprow @jordens It would be good to conclude this soon. I'd like to move on from the Fastino revison.

gkasprow, 2020-03-20:
@pathfinder49 
- [ ] lower the R48 to let's say 2k2
- [ ] connect the I2C extender directly to the EEM0_I2C (before the T6,T9)
- [ ] connect IC10 supply and all config pins to P3V3_MP
this should solve all the issues.

gkasprow, 2020-05-20:
closing, in case the issue is not fixed we will reopen it.

closed 2020-05-20 by gkasprow