sinara-hw/Fastino #51: Output noise

jordens, 2020-01-13:
# Conditions

* Kasli, EEM0 -> Fastino channels 24-31 -> IDC-BNC
* battery powered
* modules close together (that does not make any difference)
* in a shielded crate (admittedly not fully RF-tight but that doesn't make a difference)
* all DACs continuously updating (loading `channel_index << 11`) at 2.55 MS/s with 21 ns SPI period
* 1 M impedance, look at the green trace (A)
* using channel 27 which is a pretty clean one (#50) at 6.875 V output

# tl;dr:

* less than 80 nVrms/rtHz to 500 kHz
* less than 12 nVrms/rtHz above 2 MHz
* less than 4 nVrms/rtHz above 10 MHz
* nice, stable, and low noise as by design, no surprises
* the 150 Hz stuff might be something with the reference or acoustics, and might be worth investigating

## 6.9 kHz bandwidth, to 10 MHz

-106 dBV are 60 nVrms/rtHz

[image]

## 6.9 Hz bandwidth, to 10 kHz

-137 dBVrms are 54 nVrms/rtHz

[image]

## 6.9 mHz bandwidth, to 10 Hz

-140 dBVrms are 1.2 uVrms/rtHz

(coming up in a couple hours)

## model from schematics

[image]

hartytp, 2020-01-13:
Nince! Thanks for the lovely data.

hartytp, 2020-01-13:
That looks like a pleasantly good fit to the model then.

jordens, 2020-01-13:
If somebody wants to investigate the 150 Hz bump feel free. Otherwise closing.

closed 2020-01-13 by jordens

dnadlinger, 2020-01-13:
Not super important, but do you have a capture with the DACs in quiescent state as well? (I.e. does anything change apart from the clock feedthrough peak disappearing?)

dtcallcock, 2020-01-13:
I would be interesting to check the noise on the reference, both before and after the buffers (though it looks like there are only test points post buffers). 

What is the P/N of C167 @gkasprow ?  Could that be the source of microphonics?

hartytp, 2020-01-13:
> If somebody wants to investigate the 150 Hz bump feel free. Otherwise closing.

Are you confident it's not 3*50Hz line noise?

gkasprow, 2020-01-13:
C167 is generic CC0603_10UF_10V_10%_X5R
True, it could be an ultrasound from SMPS. It would affect the reference.
I would focus on C6 and C9. They are used to generate references for channels 21,22,24,25,27,28,30,31.
In Altium indexes start from 1, not zero.

jordens, 2020-01-14:
On these quiescence only removes the update feedthrough at 2.55 MHz and multiples.
The 150 Hz bump is certainly not line noise. 150 Hz line noise is very narrow in Europe and less than 10 Hz even if running on a generator.

hartytp, 2020-02-07:
I wonder if the 150Hz noise is the related to the offset trim circuitry. It's maybe a little low frequency though?

gkasprow, 2020-02-12:
This can be easily verified by removing either IC27 / IC2 or R158, R127, R140, and R220. The reference will drop to 3.9V.