sinara-hw/Fastino #53: IC3 (CC10-1212DF-E) housing not grounded

pathfinder49, 2020-02-03:
The IC3 housing is not grounded. Is this intentional? 
I'm observing a 0.6 Vp-p, ~500 kHz oscilation relative to the board ground in the IC3 housing. I found this when investigating #50.
[image]

gkasprow, 2020-02-03:
Try grounding and see how it influences the noise.

hartytp, 2020-02-03:
Indeed, there are an extra 4 pins missing from that pin-out.

@gkasprow was there a deliberate reason for not grounding it, or was this an oversight?

@pathfinder49 as Greg says, try to solder those pins to the ground plane and see if that changes the noise. If it makes things better we should look again at the other designs using this part.

jordens, 2020-02-03:
* Datsheet says grounding is optional (n.b. SMD version can't be grounded) and that it may improve coupled noise
* The case is isolated
* Therefore, depending on how you ground it and what's coupling to the case and how, doing so may also make the coupled noise worse

jordens, 2020-02-03:
It's also only two pins, not four.

gkasprow, 2020-02-03:
In isolated designs, the EMI level depends where you connect the shield. In our case we do not use isolation, so connect it to the input ground. The metal case may be capacitively coupled with transformer and grounding may help or make it worse.

pathfinder49, 2020-02-03:
Here Are some spectra for Ch31 with the IC3 housing grounded and floated. Overall it seems like grounding shifts spurs to higher frequencies. 

The casing was grounded with a wire from TP17 to one of the case pins.

[image]
[image]

hartytp, 2020-02-03:
> Overall it seems like grounding shifts spurs to higher frequencies. Overall spur power decreases.

Does it? From the data you posted it looks like not much has really changed there...

> The casing was grounded with a wire from TP17 to one of the case pins.

TP17 is really not the best place for this. Look at the layout...Greg has taken pains to create an "island" where as much of the SMPS current as possible flows to avoid common-impedance couplings (ground voltage changes due to the SMPS current). You're creating a pretty big antenna there.

As Greg suggested a few posts above, you really want to tie the shield to the SMPS input pin (look at the Henry Ott bible I gave you and he has a really nice section on why you always tie shields to the source ground for this kind of thing). In this case, that means connecting the screeing can to pin 3 of the SMPS.

pathfinder49, 2020-02-04:
I've measured the noise again with grounding to IC3 pin 3. I've also investigated the spur amplitude variability. There is no significant effect from the shielding.

pathfinder49, 2020-02-04:
I will investigate the if there is an effect when combined with the external power for #55.

pathfinder49, 2020-02-04:
Grounding the shield makes a large difference when eliminating 3V3MP noise. Unsurprisingly, the coupling to the output channels is channel dependent. 

With the shield floated and external power applied I get these spectra.
[image]

Grounding the shield, the remaining spurs in ch31 dissapear. A new ~1MHz spur appears. This coincides with the N12V0A rail starting to oscillate (fix for #54 not applied)
[image]

Applying the #54 fix The ch31 spectrum becomes
[image]

hartytp, 2020-02-04:
Interesting...what's the physics here? Why is the PSU housing related to the 3V3MP?

hartytp, 2020-02-04:
And, can you confirm how you grounded that?

hartytp, 2020-02-04:
Can I double check your y-axis scale. Is it really dBmV? (not dBm etc). If it's really dBmV then that spur is 20nV RMS. Is that correct?

Also, what's the worst-case output for a channel now? Is it much worse that this one? If so, this is starting to look excellent! Do you see any digital update noise, or is that all higher frequency?

pathfinder49, 2020-02-04:
> 
> 
> And, can you confirm how you grounded that?

The IC3 case was grounded from pin 3 of IC3

> Can I double check your y-axis scale. Is it really dBmV? (not dBm etc). If it's really dBmV then that spur is 20nV RMS. Is that correct?
>
> Also, what's the worst-case output for a channel now? Is it [...]

The y axis is dBmV (over 1 Hz). The spectrum analyser is configured to give average (as oposed to peak) bin power. Therefore I expect the spur peaks to be underestimated.

hartytp, 2020-02-04:
aah, ok. Nit pick: I would label that `dBmV/Hz` which is dimensionally correct, rather than `dBmV (Hz)` which, to me, indicates that the units for `dBmV` are Hz. For spur hunting, peak power is a better unit, although, hopefully we can kill that last spur and only have noise left!

Any idea where that last one is from?

> The IC3 case was grounded from pin 3 of IC3

Well, that makes zero sense to me.

pathfinder49, 2020-02-04:
> 
> 
> Interesting...what's the physics here? Why is the PSU housing related to the 3V3MP?

Apologies, fixed.

hartytp, 2020-02-04:
AAh, you mean that this isn't 3V3MP noise, but rather than grounding the SMPS case this way significantly improves the noise. I assume your previous measurement didn't help because the case was grounded to a TP miles away.

So, we should make sure to have an issue about grounding the case that references this data. We should keep this in mind for other designs as well

hartytp, 2020-02-04:
Where is the 3V3MP noise? Or, are you running that off a separate (clean) PSU for now?

pathfinder49, 2020-02-04:
> 
> 
> Also, what's the worst-case output for a channel now? Is it much worse that this one? If so, this is starting to look excellent! Do you see any digital update noise, or is that all higher frequency?

I've checked channels 24-31 and they are all spur free aside from the single 1.2 MHz peak. I will investigate all channels once I've finished addressing #55.

pathfinder49, 2020-02-04:
> So, we should make sure to have an issue about grounding the case that references this data. We should keep this in mind for other designs as well

See title of this issue

hartytp, 2020-02-04:
I see, so this is using a separate PSU to remove the 3V3 noise.

So, just that and the final spur to find. Good work!

hartytp, 2020-02-05:
@gkasprow based on those results, I think we should consider grounding the PSU chassis to the input ground on all designs that use this part.

At the very least, let's do this via DNPd 0R resistors to make grounding later easy. But, based on this data I'd suggest we should populate those grounding resistors by default.

jordens, 2020-02-05:
Are you certain this is the correct conclusion?
Let's disentangle the observations: I see a clean spectrum above, ok. But where is the spectrum that shows that the floating shield is still a problem with the two issues you had (at least N12V0A oscillation, P3V3MP/P12V ground bounce, and measurement issues) eliminated?
I'm certainly fine with a DNP jumper, as long as it's DNP for now.

hartytp, 2020-02-05:
> Are you certain this is the correct conclusion?

Yes, but it's not clear from the above data (particularly given that some of the descriptions were incorrect).

There are three independent observations:
- N12V0A is slightly unstable, fixed by adding a capacitor. That's one spike
- Several spurs related to grounding when Fastino is connected to Kasli (more on that later). At least one fix for this is taking Kasli from the equation and powering Fastino from floating 3V3 and 12V0 lab supplies.
- Different spurs related to the PSU chassis grounding. These can be removed by grounding the SMPS shield to its input ground pin (the previous observation that the shield grounding did not help should be ignored, since the chassis was grounded to a test point on a different part of the board)

The spurs from these sources all occur at different frequencies. We've tried several different combinations of the fixes and reproducibly are able to isolate the causes/make the spurs come and go.

Based on that, I feel pretty confident that we understand this. If you (or anyone else) have time to apply the patch and check independently that's of course always appreciated.

hartytp, 2020-02-05:
Anyway, I do agree we need a summary of this with a clear description of what setups were used and what was measured. @pathfinder49 is planning to post that later on.

jordens, 2020-02-05:
Yes. Would be great to see that description and the measurement that shows the isolated difference.
I'd be happy to reproduce it.

hartytp, 2020-02-05:
Indeed. @pathfinder49 is doing a meticulous job of this, so I'll stop getting over-excited and wait for him to post results before commenting further :)

> I'd be happy to reproduce it.

Cool! Once the write-up is done, that would be useful confirmation.

pathfinder49, 2020-02-10:
[This comment](https://github.com/sinara-hw/Fastino/issues/55#issuecomment-584073610) should really have been in this issue.

The effect of grounding the sield is summarised by this plot *(explained in the linked comment).
[image]

closed 2020-05-20 by gkasprow