sinara-hw/Fastino #54: N12V0A oscillating on some boards

pathfinder49, 2020-02-03:
I have 2 Fastino boards for testing. One of these has a 20 mVp-p, ~1MHz oscillation in the N12V0A supply. The frequency of this osscilation does not appear to coincide with other frequencies observed on the board. This noise propagates couples with different amplitudes into various output channels.

Here are some supply noise spectra for comparison.
[image]

pathfinder49, 2020-02-03:
I believe I also saw the N12V0A supply oscillating on the other board in some configurations, but have been unable to replicate this.

hartytp, 2020-02-03:
Could be the load capacitance for the LDO (see the part data sheet section on stability/capacitance). Try adding an extra 10uF ceramic right by the LDO and see if that stabilizes it.

hartytp, 2020-02-03:
How do these spurs compare with the noise on the dac  outputs?

pathfinder49, 2020-02-03:
> How do these spurs compare with the noise on the dac outputs?

Dac outputs are the bottom grouping. Lowest trace is the noise floor.
[image]

hartytp, 2020-02-03:
Ok so the worst spurs are due to the 3V3MP and the N12VOA. Interesting.

pathfinder49, 2020-02-03:
> Ok so the worst spurs are due to the 3V3MP and the N12VOA. Interesting.

I'm currently investigating one other significant spur that seems to behave differently.

hartytp, 2020-02-03:
Will be interesting to see if a larger output capacitor fixes this. C30 is a 16V X50 22uF, minimum output capacitor is 10uF so I would have expected that to be stable. It's also possible that something else funny is going on with that rail...

pathfinder49, 2020-02-03:
Adding 10uF (x5R) in parallel to C30 resolves the oscillation.

hartytp, 2020-02-03:
Sounds like a marginal capacitance then.

@gkasprow can you check for other places in Sinara we're using this LDO please?

gkasprow, 2020-02-03:
16V 22uF X5R caapcitor has much less than 10uF. X7R would be better here
[image]

dnadlinger, 2020-02-03:
@hartytp: Seems like we have been shouting helpful suggestions at @pathfinder49 in parallel (and were both right!). Do you/do we have a Sinara review checklist template somewhere we could add this to?

hartytp, 2020-02-03:
> Do you/do we have a Sinara review checklist template somewhere we could add an item for this to? (It's a common, well-known trap, but still has bitten us more than once already.)

Nope, but we should. The main wiki would probably be the best place. IIRC someone uploaded a CERN hw review checklist there. But, a shorter debugging list with the things we always tend to get wrong would be helpful as well...

gkasprow, 2020-02-03:
It's well hidden here
https://github.com/sinara-hw/sinara/wiki/SchematicsReviewChecklist

closed 2020-05-20 by gkasprow