sinara-hw/Fastino #57: Check layer stackup

hartytp, 2020-02-10:
@gkasprow can you do this please?

> Check that the layer stackup is correct and the layers weren't mixed up.

https://github.com/sinara-hw/Fastino/issues/56#issuecomment-584047456

gkasprow, 2020-02-12:
I'm not sure we can do it without breaking the board. One thing we can do is Xray. I think TS has such capabilities.

gkasprow, 2020-05-20:
Recently I tested quite a nice technology called laminography.
It's essentially kind of tomography. One can precisely check all aspects of PCB including layer spacing, via connectivity, etc.

gkasprow, 2020-05-20:
This is an example of the 3-rd layer of PCB 
[image]

closed 2020-05-20 by gkasprow

jordens, 2020-05-20:
Do you have that machine at TS/WUT/CTI?

gkasprow, 2020-05-20:
It's in another institute (https://www.itr.org.pl/en/) and offer it as a service.