sinara-hw/Fastino #58: Noise with DAC updates

pathfinder49, 2020-02-10:
I'm having a look at the performance with the Fastino DACs set to various output values.

pathfinder49, 2020-02-10:
I've configured my system using the spur-free configuration with common mode chokes from #56.

With a 1kHz bandwidth, I observing broad noise peaks around ~4.6 MHz and ~9.2 MHz. These are present with a range of DAC voltages. 

Up to 5 MHz scan
[image]

5-10 MHz scan
[image]

hartytp, 2020-02-10:
A few questions here...

1. These plots look rather different to the data you posted [previously](https://user-images.githubusercontent.com/14295481/74144431-9d37ba80-4bf4-11ea-8d7e-7304c42ced9b.png). The noise floor seems quite a bit higher in the new data (for the same measurement BW). Also, comparing to the orange trace in the old data there are fewer spurs, but I guess that's because the old narrow-band spurs are swamped by a broad noise peak at around 4.75MHz? From memory, it the spectra you posted above feel cleaner than the ones we were looking at last week (lots of spurs around 6-7MHz). Is that because you moved the equipment to a new (quieter) lab?
2. Are the DACs updating continuously in this? I'm surprised not to see a big peak at the DAC clock frequency.

pathfinder49, 2020-02-10:
Unlike my measurements in #56, Kasli and Fastino were placed in a floated rack. I observe the following:
*  Breaking the contact between the grounded Fastino front panel and the rack removes the bumps. 
*  Breaking the contact between the floated Kasli front-panel and the rack makes no difference.
*  Adding an additional ground strap to the Rack removes the bumps.

By eye, 5-10 MHz noise performance is better with Fastino in contact with the rack and an additional grounding strap.

pathfinder49, 2020-02-10:
Playing around with this, a ground strap to Kasli-ground also appears to resolve this. However, I now have trouble reproducing the broad previous spectrum.

pathfinder49, 2020-02-10:
> A few questions here...
>
> 1. These plots look rather different to the data you posted [...]

1. The previous spurs may have been pickup in the previous building. I have not observed these in the new location. Whenever the broad peaks are eliminated, I observe no high frequency spurs.

2. The LDAC (active low) is low and CLR (active low) is high. So the DACs should be updating.

pathfinder49, 2020-02-10:
With Kasli and Fastino ground tied to the analyser ground I get this noise spectrum.

[image]

hartytp, 2020-02-10:
Is the clock line toggling?

hartytp, 2020-02-10:
Also I’m somewhat suspicious of you y-scale here. Is that really dBV/kHz? If so the analyser noise floor would be 300nV/rtHz which seems very high. Unless I’m missing something the units are either off or the analyser is configured to give a very high noise floor (or its much noisier than I expect).

hartytp, 2020-02-10:
For the close-in noise floor Robert measured -106dBV in 7kHz BW which is inline with expectations.  You’re posting -85dBV in a 1kHz bw which doesn’t seem right. I suspect you’ve got the analzyer config wrong/aren’t using the units you think you are.... silly question but does the data you save match the data you see on the screen? I.e. are you working in noise densities on the analyser but saving data in integrated power or something like that (although even then I’m not sure this would totally make sense)

hartytp, 2020-02-10:
In some of your other posts I see similar numbers quoted as being in dBmV/Hz. -85dBm/Hz would be -105dBV/kHz which Is more in line with expectations...I don’t think this changes any conclusions but it’s better to get the units right

pathfinder49, 2020-02-10:
> 
> 
> In some of your other posts I see similar numbers quoted as being in dBmV/Hz. -85dBm/Hz would be -105dBV/kHz which Is more in line with expectations...I don’t think this changes any conclusions but it’s better to get the units right

The analyser gives spectra in units of dBmV over the band-width. I supect this is what happend, but will check tomorrow.

jordens, 2020-02-11:
>  -85dBm/Hz would be -105dBV/kHz

In a 50 Ohm system -105 dBV is -92 dBm and -105 dBV in 1 kHz is -122 dBm in 1 Hz.

pathfinder49, 2020-02-11:
I'messed up my dBmV to dBV conversion. I subtracted 30 instead of the correct 60 dB difference.

pathfinder49, 2020-02-11:
> Is the clock line toggling?

The DAC SCLK is not toggling.

hartytp, 2020-02-11:
> In a 50 Ohm system -105 dBV is -92 dBm and -105 dBV in 1 kHz is -122 dBm in 1 Hz.

Oops, adding to the confusion, that `dBm` was meant to be `dBmV`.

hartytp, 2020-02-11:
> I messed up my dBmV to dBV conversion. I subtracted 30 instead of the correct 60 dB difference.

I thought it was something like that. Well...oops...anyway at least with that factor all the numbers look consistent.

hartytp, 2020-02-11:
> The DAC SCLK is not toggling.

Well in that case the DACs aren't updating. Explains why we don't see any spurs ;)

closed 2020-02-13 by pathfinder49