sinara-hw/Fastino #59: Radiative Noise from Kasli

pathfinder49, 2020-02-10:
In many configurations conductive noise from Kasli dominates.In some configurations spurs matching the Kasli switch-mode supply can still be seen once conductive spurs have been eliminated.

Using the "Spur free" configuration (with CMCs in Fastino outputs) from #56 I note the following:
*  There are spurs Spurs when Fastino is placed adjacent to kasli (to the right in my configuration).
*  These spurs are present even if neither Kasli nor Fastino front-panels are in contact with each other or the crate.
*  These spurs rapidly drop off with separation. With a 2HP gap the spurs are just above my noise floor.
*  These Spurs are suppressed by adding a ~1 mm thick, floating aluminium shield between Kasli and Fastino.

[image]

pathfinder49, 2020-02-10:
I'd like to add that I don't see this as a problem. However, it is worth noting.

jordens, 2020-02-10:
Ah. And I assume that's with a ribbon running from Kasli EEM0 across the Fastino underside to Fastino EEM0? Is it Kasli or the ribbon near the Fastino channels? The latter could explain a few things.

hartytp, 2020-02-10:
I believe the coupling is directly from Kasli based on previous experience but agreed that it would be good to sanity check by moving the ribbon cable from one side to the other of the screening panel

pathfinder49, 2020-02-11:
I see no difference from moving the ribbon cable or changing the cable routing.

pathfinder49, 2020-02-11:
Placing Fastino on the other side of Kasli (adjacent to the left), I also see some switch-mode spurs. In this configuration the ~220 kHz sub-harmonic dominates. I also see a small, but detectable ~450 kHz spur.

hartytp, 2020-02-11:
So the conclusion is that this is that in this particular case, the spurs are a direct radiate coupling between the boards and not related to the EEM. That feels consistent with the previous measurements we've made.

Closing this issue since it's more of an observation than an issue requiring action...

closed 2020-02-11 by hartytp