sinara-hw/Fastino #60: Channel Spur Sensitivity Overview

pathfinder49, 2020-02-11:
@hartytp has requested that I produce a table showing spur powers across the Fastino channels. This may be useful in identifying layout improvements for #57. 

With reference to the different spur sources summarised in #56, which spurs would be of particular interest?

pathfinder49, 2020-02-11:
I've had a preliminary look at the ~220 kHz and ~450 kHz Kasli switch-mode spurs resulting from using a non-floated PSU with choke in the Fastino output. (No grounding strap is used and Fastino is not in contact with a crate.)

Below I tabulate the visible spurs and their approximate amplitudes. 
*  As I only have defective IDC to BNC adapter boards available. I'm unable to easily test channels 6, 14, 22 and 30.
*  I found these spur powers to be hard to reproduce. Re-plugging Kasli can change these spur powers by over 10 dB. 
*  These measurements were taken with additional capacitors added around FL1 on Kasli. the ~900 kHz spur is therefore suppressed. Fixes for #53 and #54 have been applied. 

I'd like to draw particular attention to channels: 15, 21, 26 and 27.

I estimate relative spur powers to have an error of ~1.5 dB.

| Channel | Spur frequencies /kHz | Amplitudes /dBmV |
| --- | --- | --- |
0 | 220, 450 | -39, -44
1 | 220, 450 | -40, -45
2 | 220, 450 | -38, -41
3 | 220, 450 | -39, -41
4 | 220, 450 | -37, -40
5 | 220, 450 | -41, -45
6 |  |  |
7 | 220, 450 | -43, -45
8 | 220, 450 | -40, -43
9 | 220, 450 | -37, -39
10 | 220, 450 | -40, -44
11 | 220, 450 | -41, -43
12 | 220, 450 | -37, -38
13 | 220, 450 | -36, -40
14 |  |  |
15 | 220, 450 | -35, -38
16 | 220, 450 | -39, -43
17 | 220, 450 | -40, -45
18 | 220, 450 | -37, -40
19 | 220, 450 | -37, -40
20 | 220, 450 | -39, -43
21 | 220, 450 | -35, -36
22 |  |  |
23 | 220, 450 | -41, -44
24 | 220, 450 | -38, -42
25 | 220, 450 | -40, -43
26 | 220, 450 | -44, -50
27 | 220, 450 | -35, -37
28 | 220, 450 | -39, -44
29 | 220, 450 | -41, -45
30 |  |  |
31 | 220, 450 | -41, -45

hartytp, 2020-02-11:
Thanks for that! So on the 220kHz spur the channel-channel variation is a factor of a few (a bit under 10dB), which doesn't feel that surprising.

Even on the 450kHz the peak variation is something like a factor of 4.5 (13dB). I'm not sure how surprised to be by that. It's maybe a little on the high side, but nothing I'm unduly worried about.

hartytp, 2020-02-11:
The one interesting observation from a quick skim over the above is that generally the best channels look like the ones closest to the EEM connector while the worst channels appear to be the ones near the front.

Channels on the row closest to the EEM connector are: 0, 2, 4, 6, 9, 12, 15, 18, 21, 24, 27
Channels driven by ref buffer IC19 are: 0, 2, 4, 6, 9, 12, 15, 18
Channels with spurs worse than -42dBmV are: 2, 3, 4, (6?), 9, 12, 15, 18, 19, 21 (!) , 27

Best channels are 26 and 17, both of which are close to the FPGA.

@pathfinder49 do the spur heights change with the set DAC output voltage (if so they would be on the reference/reference ground)?

hartytp, 2020-02-11:
Anyway, not really clear to me what conclusions we can usefully draw from that...

pathfinder49, 2020-02-11:
The above data does not represent general spur senstivity. Notably channel 27 looked fairly clean in general iirc. However, it performs worst in this measurment.

pathfinder49, 2020-02-11:
I'd really need an indication what spurs you would like to understand. Different spurs behave differently across channels.

hartytp, 2020-02-11:
< Notably channel 27 looked fairly clean in general iirc. However, it performs worst in this measurment.

Do you mean that in different measurements (different grounding configurations?) that was a "good" channel? i.e. which channels are good/bad is very configuration-dependent, possibly pointing to several different coupling mechanisms which can dominate depending on the overall system configuration?

One thing, are those actual spur amplitudes or are we in some funny noise units? e.g. can I read from -30dBmV that the spur is ~30uV RMS into the 50Ohm load?

pathfinder49, 2020-02-11:
The analyser was configured in "peak" detector mode. The values should therefore be spur amplitudes as you describe.