sinara-hw/Fastino #62: Digital Crosstalk Summary

pathfinder49, 2020-02-20:
Fastino 1.0 has significant digital crosstalk when DACs are updated. Updating a single channel can cause digital crosstalk spurs over -40dBmV in other channels.

Most of the crosstalk I have observed can be explained by the mechanisms detailed below.

## Configuration Details
Unless otherwise specified, the configuration was as follows:
* A single DAC channel updating at 2.55 MS/s. The DAC word is chosen to be 0x0000 (-10 V).
* Look at the 2.55 MHz spur using a 50 Ohm impedance spectrum analyser (noise floor: ~ -85 dBmV)
    * Using a BNC breakout board with CMCs filtering the DAC-putput
* Spur free configuration from #56 with a grounding strap between Kasli/Fastino ground and the analyser ground. (Though this doesn't appear to matter here)
* R66 and R67 populated in each Fastino channel. (No CMC on the Fastino board)
[image]

# Digital Trace Coupling
 Channels with analogue vias near updating digital traces experience crosstalk. The examples below demmonstrate the following points:
* Coupling occurs through vias. Coupling to traces or pads is not dominant.
* When the CMC slot on fastino is not populated the negative output vias are not sensitive. These vias become sensitive if a CMC is placed in the intended slot.

## Example: Digital Traces Couple to Analogue Vias
Updating channel 28 produces digital crosstalk to channel 11. The channel 28 DAC word was chosen as 0x0000 (-10 V). This keeps the DIN line static. The relevant trace routings are shown in the image below.
[image]
The updating digital traces (SCLK and CS) run underneath the output side pad of R66. Meanwhile, the the analogue via close to the traces in on the DAC side of R66.

**Removing R66 results in the -50 dBmV digital crosstalk spur becoming undetectable.**

This behaviour is not generally the case. Consider the crosstalk of channel 22 updates to channel 14:
[image]
Removing R66 here causes the spur to increase from -49 dBmV to -45 dBmV.

This behaviour is inconsistent with coupling to pads/ traces. However, it is consistent with coupling to the anaogue vias.

## Example: DAC OUT_P vs OUT_N Crosstalk Susceptibility
SCLK and CS trace routing of channel 22 relative to channels 14 and 17 is very similar and a good basis for comparrison.
[image]
The traces are only near to the OUT_N via of channel 17. Meanwhile, SCLK runs very close to the OUT_P via of channel 14. 
The two channels experience very different crosstalk of -82 dBmV and -49 dBmV respecively.

This behaviour seems to be related to the effective grounding of OUT_N. The table below shows that adding impedance between the OUT_N via (close to digital lines) and ground reults in significant spurs.
| Configuration | Ch17 spur /dBmV |
| --- | --- |
| R66 & R67 populated | -82 |
| **OUT_N floated, R66 populated** | **-50** |
| OUT_P floated, R67 populated | -86 (resolved with averaging) |
| **OUT_N & OUT_P floated** | **-58** |
| **R66 & R67 removed, CMC populated** | **-63** |
| R66 & R67 removed, air wires from CMC pads | -83 |

# Digital Trace Termination (Common Impedance?) Coupling
Channels in the vicinity of the updating DAC tend to experience crosstalk. This appears to be mostly due to (common impedance?) coupling from the digital trace termination. The example below demmonstrates this point.

## Example: Termination Coupling to Nearby DACs
Channel 4 updates result in digital crosstalk to many channels. The majority of these channels are not near the digital traces for channel 4.

The relevant channels and traces are shown in the image below.
[image]

To determine the effect of (common impedance?) coupling to the digital trace termination, I removed the Termination resistors and capacitors on channel 4. The table (below) demostrates that removing the termination removes most of the coupling to channels without vias close to the digital lines.

*Aside: Channel 4 still worked correctly without termination and I did not observe output errors when continuously updating Channel 4.*

Channel\Configuration | DIN toggling (-9.5 V), Ch4 terminated | Static DIN (-10 V), Ch4 terminated | DIN toggling (-9.5 V), Ch4 **NOT** terminated | Static DIN (-10 V), Ch4 **NOT** terminated
-- | -- | -- | -- | --
Ch1 | -39.5 | -40 | -39 | -40
Ch3 | -62 | -82 | -66 | gone
Ch5 | -74 | -76 | gone | gone
Ch6 | -80 | gone | gone | gone
Ch7 | -69 | -69 | -77 | gone
Ch10 | -75 | gone | gone | gone

# Notes
* The examples above focus on the vias around the CMC position. Crosstalk also occurs if digital traces are routed close to analogue op-amp vias.
* I did not see any instances where using a CMC on Fastino gave an advantage over using a CMC on the BNC breakout board. I did encounter cases where using a CMC on Fastino resulted in worse digital crosstalk.

gkasprow, 2020-02-22:
Note, that removal of termination reduces the effects of inductive coupling between traces but does not affect the capacitive coupling. The parallel termination simply injects current nearby the chip. It may then propagate via analog part somehow. So, removal of the termination solves one issue but may create other ones.

hartytp, 2020-02-27:
> Note, that removal of termination reduces the effects of inductive coupling between traces but does not affect the capacitive coupling. The parallel termination simply injects current nearby the [...]

What issues could removing the AC shunt termination cause (beyond SI) and how would we test for it? We will measure the SI with a low capacitance high-bandwidth probe. Are there any other tests we should be doing?

pathfinder49, 2021-02-09:
Digital to analogue crosstalk is significantly improved in Fastino v1.2 (see #85)