sinara-hw/Fastino #63: Remove DAC Digital Trace Termiantion

pathfinder49, 2020-02-21:
The termination degrades Fastino performance and is not needed. See below:
* The termination of the digital traces to the DACs results in digital crosstalk (see #62). 
* Empirically, the termination is not required for correct DAC opperation.
* The longest digital traces on Fastino are ~10 cm long. With an approximate signal propagation speed of [15 cm/ns](https://www.edn.com/rule-of-thumb-3-signal-speed-on-an-interconnect/) I would expect at 1 GHz or faster to require termination. The Fastino SPI frequency is only 50 MHz.

gkasprow, 2020-02-22:
Did you check operation without termination? Did U observe the clock edges with low capacitance probe? 
It's not frequency but the signal rise time that matters. Even with 1Hz and LVPECL edges, you will experience issues with lack of termination :)
I added termination because I could not fit the series termination close to the CPLD.

pathfinder49, 2020-02-24:
I've checked the DAC operates fine without termination on channels 4 and 27 (longest digital trace).

Here is a comparrison of the clock line with and without termination. I measured this using a 9.5 pF, 10:1 passive probe and a high impedance scope.
| SCLK terminated (ch24) | SCLK un-terminated (ch27) |
| --- | --- |
| [image] | [image] |

These both look acceptable to me.

hartytp, 2020-02-24:
@pathfinder49 BTW that scope has a web interface that lets you get screen grabs directly without having to use your camera phone. 

Can you set cursors on the logic thresholds specified in the DAC data sheet please?

pathfinder49, 2020-02-24:
Updated my post as requested

pathfinder49, 2020-02-24:
> Can you set cursors on the logic thresholds specified in the DAC data sheet please?

The schematics show VLOGIC connected to the P5V4 plane. Measuring on the board confirms VLOGIC being P5V4.

There are 2 different statements in the data-sheet on logic levels.
* On page 4 VINL and VINH are given as 0.8 and 2.4 V respectively. Fastino meets this spec.
* On page 6 different digital lines require VINL/VINH of 10%/90% of VLOGIC or VDD. Fastino does not meet this spec.
  * SCLK: VINL/VINH of 10%/90% of VLOGIC
  * DIN: VINL/VINH of 10%/90% of VDD

Am I missing something or are the Fastino DACs being driven outside spec?

dnadlinger, 2020-02-24:
They also spec hold timings for low 0V/high 3V in the table, so presumably that's also valid. (The hold time is a bit longer for that, which makes sense.)

dnadlinger, 2020-02-24:
(Requiring anything but "comfortably" below 0.8V/above 2.4V with some allowance for PVT changes would be very unusual for these kinds of slow-ish serial interfaces. While issues like this in the analog part would definitely be worth worrying about, I'd not expect any issues in the digital logic here.)

gkasprow, 2020-02-25:
@pathfinder49 There is nothing wrong with the logic levels. The page 4 states the requirements while the page 4 defines the measurement conditions.

gkasprow, 2020-02-25:
It looks like the ICE40 has high-enough output impedance. It is not specified in the datasheet. I extrapolated it to be something like 20Ohm, but it looks like it was higher and indeed, it is sufficient as a series termination. 9pf of scope capacitance is quite a lot. I have sub-pf probes but don't have any Fastino to look at it. 9pF of capacitance gives Z=117Ohm at 3-rd harmonic, so these plots are not representative. On the other side, the ADC has 10pf capacitance...

hartytp, 2020-02-25:
@pathfinder49 which probe are you using? The 500MHz 10:1 passive probes that go with the MSO should be much less than that.

hartytp, 2020-02-25:
Also remember to be careful to minimise ground inductance here—don’t use a croc clip, use a sprig clip to ground to somewhere a mm or two from the probe tip

dnadlinger, 2020-02-25:
> The 500MHz 10:1 passive probes that go with the MSO should be much less than that.

Nope, those are ~10 pF, like most 10:1 probes (and, to come to Marius's defense here, I don't think we have any nice active probes).

hartytp, 2020-02-25:
Ok! I stand corrected, I thought it was lower. The other option is to fall back to soldering a 500R SMT onto the pad as a probe tip and then solder the end to some coax with the shield pig tailed and soldered to the ground plane.

But we should but a fet probe...

pathfinder49, 2020-02-25:
> 
> 
> Also remember to be careful to minimise ground inductance here—don’t use a croc clip, use a sprig clip to ground to somewhere a mm or two from the probe tip

I soldered a pin onto the CMC ground pad of ch27 and clipped the scope ground onto that.

gkasprow, 2020-02-26:
The easiest low capacitance 1:10 probe is 450R resistor (2x 910R 0603) in series with a piece of 50Ohm coax cable. Such a probe gives very low rise time and keeps the same 500R impedance up to the GHz range.
One can also use a higher value resistor.

pathfinder49, 2020-03-04:
@gkasprow you [reccomended series termination](https://github.com/sinara-hw/Fastino/issues/66#issuecomment-592187275). What resisor valus would you use? I've estimated the trace impedance as ~50 Ohm. However, I'm struggling to find values for the FPGA output imedance. 

Also, CLRn is a multi-drop bus. Do you believe series terminating this at the FPGA will be adequate?

pathfinder49, 2020-03-04:
I've now got a 1 pF, 1 GHz probe. There is a clear difference between terminated and un-terminated performance.

| SCLK terminated (ch24) | SCLK un-terminated (ch27) |
| --- | --- |
| [image] | [image] |

hartytp, 2020-03-04:
Nice! If that’s the worse channel I don’t see any evidence of a si issue there (not near a double triggering)

@gkasprow the overshoot is ~1V do you think that will lead to worse digital feed through (eg when the diodes conduct?)

I guess the answer to my question is “no” since @pathfinder49 has measured the noise and checked that it gets better not worse when we dnp the termination.

Based on these measurements does everyone agree we should dnp the shunt terminations but consider adding a small (10R?) series termination at the FPGA?

gkasprow, 2020-03-04:
Be careful with 1V overshoots because they can open the protection diodes and cause current peaks that may interfere with circuits that were not designed to act like the clamping circuits. For example, old PCI used such an approach. The best series termination is probably something between 10 and 20 Ohm. 
WE don't care about CLRn since it is an asynchronous interface so any topology will work.

hartytp, 2020-03-05:
@gkasprow why do you think the transients aren't clipped to -0.3V by the diodes? I don't see any evidence of clipping in the data that @pathfinder49 posted. Do you think it's a measurement artifact (e.g. ringing in the probe due to ground inductance?)

hartytp, 2020-03-05:
NB Vdd is 5V4 for the DAC, so only the negative transients could prove problematic. 

In any case, it sounds like a good plan for the next revision is:
- [ ] DNP the termination
- [ ] Add 10R series termination close to the FPGA (we will optimize the resistor values once the hw arrives)
- [ ] when re-routing the digital traces try to keep all digital lines as short as possible

jordens, 2020-03-05:
Overshoot or undershoot that are outside the spec are a massive problem. We can't sell or support systems that knowingly and intentionally exceed specs for no good reason. Whether there is added noise or activated clamping diodes or not doesn't matter.

jordens, 2020-03-05:
To state that "the termination is not required" is also speculative and potentially very wrong, since for example there hasn't been any analysis of the effect of unterminated bounce on the FPGA.

hartytp, 2020-03-05:
@jordens sure. The plan is to add source termination in the next revision and take more measurements before making any decisions.

I'm happy to keep the default population option being AC shunt terminations populated and series terminations 0R for the next revision to minimize the chances of breakage. 

> Whether there is added noise or activated clamping diodes or not doesn't matter.

The point I was making to Greg wasn't that we shouldn't ignore the bounce, but rather that the lack of clamping could be indicative of a measurement error (in which case we might want to take a bit more data before making any decisions). I don't have well-calibrated intuitions here so am discussing what to do with @gkasprow

jordens, 2020-03-05:
> lack of clamping

AFAICS clamping is consistent with what you measure. Given the ringing period, probe impedance and bandwidth I would not expect a clamped waveform to look much different.

hartytp, 2020-03-05:
One other thought here...

The aim of the shunt termination is to suppress ringing at frequencies where the wavelength becomes comparable to the trace length. However, our main concern is frequencies close to the filter cut-off; high frequency components can easily be filtered out later if the application is sensitive to them.

This From the data @pathfinder49 posted, the ringing period looks to be approx 4ns, so 250MHz. However, the pole in the current termination is 16MHz. @pathfinder49 can you see what the ringing looks like with a 10pF termination capacitor? That should buy 20dB on the spurs due to the termination. It may be possible to reduce the capacitor further/eliminate it once we add series termination, but we'll want to measure that.

pathfinder49, 2020-03-05:
> The aim of the shunt termination is to suppress ringing at frequencies where the wavelength becomes comparable to the trace length. However, our main concern for digital-analog cross-talk is frequencies close to the filter cut-off; high frequency components can easily be filtered out later if the application is sensitive to them.

What's your model for this? I think it's quite likely that the termination related crosstalk couples into other channels after the filter.

hartytp, 2020-03-05:
> Do you think it's a measurement artifact (e.g. ringing in the probe due to ground inductance?)

Answering my own question, I don't think it is a measurement artifact because of the data that @pathfinder49 posted with the shunt termination, showing no significant ringing.

> AFAICS clamping is consistent with what you measure. Given the ringing period, probe impedance and bandwidth I would not expect a clamped waveform to look much different.

Te probe 3dB bandwidth is 1.5GHz, which is quite a bit higher than the ringing frequency. The impedance is 1pF which is relatively small compared with other capacitances around.

Anyway, I would have expected clipping to have introduced some asymmetry since clipping only occurs on the low logic level not the higher one.

hartytp, 2020-03-05:
> What's your model for this? I think it's quite likely that the termination related crosstalk couples into other channels after the filter.

I think we're talking about different things...

The termination is AC-coupled and is not really intended to terminate the 2.5MHz components you're worrying about (the wavelength is so long there we won't get ringing). It really only needs to terminate the components at hundreds of MHz that you can see causing issues. By decreasing the capacitance we should still be able to provide a good termination for the frequencies where it's needed, but significantly reduce the leakage at the frequencies you care about.

Put it another way: if we have some very low-level 250MHz noise on our analog outputs we won't care (a) because our trap RC filters will remove it (b) because it's so far from any motional mode frequecy.

jordens, 2020-03-05:
> 3dB bandwidth is 1.5GHz, which is quite a bit higher than the ringing frequency. 

It says 1 GHz above. With 300 MHz ringing, you have just the third harmonic within in your bandwidth. I am unsure how visible the distortion of ringing due to clamping will be given those constraints.

> I would have expected clipping to have introduced some asymmetry 

It is asymmetric. The undershoot is visibly smaller than the overshoot.

dhslichter, 2020-03-05:
There are usually IBIS models for both FPGA (lets you determine output impedance, and thus optimal series termination) as well as DAC (lets you see IV curves for overshoot/undershoot and gives a sense of how bad clipping is from protection diodes), would be worth looking at.

Is there a rise time option on the FPGA IO config for these channels?  Sometimes there is (there is on Kintex 7, pretty sure Artix 7 is the same), and we can turn it to slow (instead of fast, if it is currently defaulting to fast) to reduce the power of the high-frequency components in the digital signal.

dhslichter, 2020-03-05:
@pathfinder49 you should also measure the digital signal both at the DAC and at the FPGA to check for ringing/overshoot.  The overshoot you have shown above certainly looks nasty, I agree with @jordens that we don't want to intentionally design something that has overshoot like that.

hartytp, 2020-03-05:
Measuring at the FPGA isn't easy right now since there aren't convenient places to probe. We'll revisit this all again in the next revision when there are series termination resistors at the FPGA which we can probe.

pathfinder49, 2020-03-05:
> you should also measure the digital signal both at the DAC and at the FPGA to check for ringing/overshoot.

I've thought of this. However, there is no non-destructive way for me to measure the digital traces close to the FPGA.

pathfinder49, 2020-03-05:
> The overshoot you have shown above certainly looks nasty
I agree the overshoot from the high to low transition to -0.7V is concerning and out of spec. (Low to high is fine as VLOGIC=5.4 V).

gkasprow, 2020-03-05:
The FPGA has vias. You can scratch one and stick the probe.
ICE40 has 8mA IOs. I used IBIS to find right termination impedance, but I think it was the trace Z that was the most important.
The R is not specified as a number but as an I=f(u) characteristics.
We can estimate R= dv/di
Let's take Vcc=1.2V +/- 500mV , the mid-range operating point and typical current
Pulldown: 
R= (1.7V - 0.7V) / (33.8mA-21.6mA) = 1/12.2 = 81Ohm
Pullup: 
R=(1.7-0.7)/(37-23.4) = 1/13.6 mA = 73Ohm

gkasprow, 2020-03-05:
I took data from lvc250io standard in the[ IBIS model](http://www.latticesemi.com/view_document?document_id=48057)

pathfinder49, 2020-03-06:
I've dead-bugged a few different ac termination capacitors onto channel 27 (longest digital traces). You can see the the SCLK wave-forms (measured at the DAC) below.

All treminations consist of a capacitor and 100 Ohm resistor in series
| Capacitacne | Waveform |
| --- | --- |
100 pF (currently used on Fastino, measured on ch 24) | [image]
10 pF | [image]
 6.8 pF | [image]
3.3 pF | [image]

This looks like we can safely use a 10 pF capacitor (with a 100 Ohm resistor). I would expect this 10x capacitance decrease to result in 20dBV reduction in cross-talk from the digital termination.

pathfinder49, 2020-03-07:
> The FPGA has vias. You can scratch one and stick the probe.

Here is the SCLK line at the FPGA when operating **without** termination. 
[image]

This is within the FPGA spec. Series-termination should therefore work fine.

pathfinder49, 2020-03-07:
We have two options for reducing crosstalk from the termination:
1. Reduce the AC termination capacitor to 10 pF. This should give ~20 dBV crosstalk reduction at frequencies of a few MHz.
2. Switch to using series termination. This should eliminate all termination crosstalk.

Option 2. seems like the better solution. Unless there are objections, I will implement series termination.

hartytp, 2020-03-07:
> I've dead-bugged a few different ac termination capacitors onto channel 27 (longest digital traces)

Dead-bugged, or just soldered them onto the pads?

hartytp, 2020-03-07:
> This looks like we can safely use a 10 pF capacitor (with a 100 Ohm resistor). For few MHz frequencies, the 10x capacitance decrease should result in a 20dBV reduction of digital cross-talk from the termination .

What's the undershoot (min negative value measured) with the 10pF capacitors? For the next revision we should aim to keep the worst-case undershoot to ~200mV to give us some margin. I'd guess that will be something in the 10-20pF range. Once we have the new hw with pads for series termination we can quickly have a play and find something closer to optimal...

hartytp, 2020-03-07:
> Option 2. seems like the better solution. Unless there are objections, I will switch to series termination.

Do we win anything by removing the pads for the shunt terminations?

If it's not a problem for the layout, I'd prefer to have the option for both shunt and series termination in the next revision. Maybe go for 20pF shunt and 0R series. Then order some hw to play with, with the aim of optimizing values for the v1.3 hw.

IIUC, in the current hardware it's hard to know how much impact the shunt termination has on the noise since the "spur floor" is set by the vias/routing. Once we fix this we should be able to get better data and make an informed decision.

pathfinder49, 2020-03-07:
> Do we win anything by removing the pads for the shunt terminations?

Space, though not a major concern. The current 0201 capacior size is not very convenient for reworking the board. With 0603 caps, the termination becomes much larger.

>  Maybe go for 20pF shunt and 0R series.

Why not use series termination by default. As I understand it, the only downside of this is that it does not give a nice waveform at the FPGA. However, the reflections to the FPGA are withion the FPGA spec.

> it's hard to know how much impact the shunt termination has on the noise since the "spur floor" is set by the vias/routing.

This is not correct. The current termination causes crosstalk-spurs of ~-70 dBmV.

hartytp, 2020-03-07:
> Space, though not a major concern. The current 0201 capacior size is not very convenient for reworking the board. With 0603 caps, the termination becomes much larger.

Well, if we have space to spare then it's not a concern. If we're pushed then we should consider this more seriously.

> Why not use series termination by default? As I understand it, the only downside of this is that it does not give a nice waveform at the FPGA. However, the reflections to the FPGA are within the FPGA spec

hartytp, 2020-03-07:
> Why not use series termination by default? As I understand it, the only downside of this is that it does not give a nice waveform at the FPGA. However, the reflections to the FPGA are within the FPGA spec

Because what we have currently works pretty well. The series termination should work fine, but will need testing. If we leave both options open then we know that after a quick bit of testing we'll be able to find something that works nicely. We may also find that a combination of series and shunt termination provides the best overall solution.

> This is not correct. The current termination causes crosstalk-spurs of ~-70 dBmV.

Ok. Thanks for the reminder. How much lower was it without termination?

pathfinder49, 2020-03-07:
> Ok. Thanks for the reminder. How much lower was it without termination?

If there is no other significant coupling-mechanism, the termination-spurs became undetectable (measurement noise floor -85 dBmV)

hartytp, 2020-03-07:
So we have no evidence for termination making more than a factor of ~4 difference to the spur. That suggests that reducing the capacitor to 20pF should reduce the spur to our noise floor while still providing an excellent termination. 

If the series termination works well we can completely remove the shunt termination in a future revision but let’s keep it for now unless we find a good reason to remove it.

hartytp, 2020-03-07:
Put it another way: don’t underestimate how much of your life it will take to add and route all those series terminations and then test to make sure all works. If you can get the same effect by just changing cap values then surely that’s what we should do...

pathfinder49, 2020-03-09:
> reducing the capacitor to 20pF

I've updated my post with a clearer 10 pF trace and a 15pF trace.

Measuring forr a minute, 10pF gives a minimum of -0.295 V. This is within the DAC spec.
15 pF gives a minimum of -0.135 V. This leaves plenty of margin to the DAC spec.

hartytp, 2020-03-09:
> Measuring forr a minute, 10pF gives a minimum of -0.295 V. This is within the DAC spec.
15 pF gives a minimum of -0.135 V. This leaves plenty of margin to the DAC spec.

Sounds good. Let's go for 15pF.

pathfinder49, 2020-03-11:
I've tired different series terminations by cutting the ch31 SCLK trace close to the FPGA. I then bridged the gap with different resistors. Parallel termination has been removed.

Series termination works well. From my measurements, I'd choose a series resistance of ~80 Ohm.

| 0 ohm | 51 ohm | 82 ohm | 100 ohm |
|---|---|---|---|
[image] | [image] | [image] | [image]| 

Based on these results, I will use 80 ohm series termination and DNP the parallel termination.

hartytp, 2020-03-12:
Nice job @pathfinder49 !

The other thing one needs to be aware of with series termination is the signal rise time, since you're now creating an RC filter with the trace capacitance. I don't see a specification for that in the data sheet, but the 51Ohm trace you post certainly looks fine to me. What was the undershoot in that case?

@gkasprow what do you think? This seems somewhat higher than the series termination values you expected from your calculation...

Anyway, DNPing the shunt termination pads and adding a ~50Ohm series termination seems like a relatively safe bet for the next revision. We should consider only stuffing one board initially and double checking the SI before stuffing the rest.

gkasprow, 2020-03-12:
making 80Ohm traces can be hard for existing stack up.
Having such high series resistance makes RC filter instead of properly matching the trace impedance. With such resistance ( which makes it over 100Ohm with FPGA pin impedance) makes the rising time longer so the impedance matching is not necessary for such trace length
But let's do whatever works.

hartytp, 2020-03-12:
@gkasprow Indeed, this is likely more RC filtering than termination. I suspect it will have a pretty similar effect to just decreasing the FPGA drive strength (although, IIRC we can't do that on these FPGAs, can we?). 

Am I right that from the numbers you posted above, the transmission line is decently source terminated by the FPGA?

But, if you're happy with the rise-time for those traces, is there any reason that the series resistor wouldn't be an acceptable solution to the problem?

pathfinder49, 2020-03-12:
> 51Ohm trace you post certainly looks fine to me. What was the undershoot in that case?

about 150 mV. From my measurements, resistors smaller than ~70 ohm have some overshoot. Resistors larger than ~80 ohm get a 2 step rising/falling edge. 

> The other thing one needs to be aware of with series termination is the signal rise time

I'm inclined to err on the side of a higher resistance as:
* For DIN the hold time is specified as 5 ns. We're comfortably above this.
* The trace impedance is not controlled. If the trace impedance is higher in some batches, a higher resistance would avoid overshoot.

> DNPing the shunt termination pads and adding a ~50Ohm series termination seems like a relatively safe bet for the next revision.

I'm using 82 ohm, 0201 resistors for the series termination (space constraints close to the FPGA). I will keep parallel termination, but not populate the 0603 capacitors. This should allow fairly easy switching to parallel termination if required.

gkasprow, 2020-03-12:
sounds like a good plan

hartytp, 2020-03-12:
> For DIN the hold time is specified as 5 ns. We're comfortably above this.

You haven't demonstrated that AFAICT. That would require a simultaneous measurement of both the clock and data lines. Trigger the measurement from the clock and look at the "eye" for the data.

Looking at the clock doesn't tell you everything. e.g. about skew between the data and clock lines, or any timing jitter between the two.

hartytp, 2020-03-12:
Also, there are generally requirements for minimum edge rate at the transition. Slow edges can cause issues with the sampling logic. This DAC doesn't specify the max signal rise time, but it is likely still sensitive to it (any ideas @gkasprow about what we should aim for?).

Anyway, I'm fine with sticking the pads in for the series resistor (since you're using a combined resistance, including the FPGA drivers, that's larger than the trace impedance, it's not really matching as much as filtering). DNP the shunt termination and take more data once the next hw revision arrives.

NB since the trace capacitance varies significantly between channels, we'll need to re-check on more channels on the new hw.

pathfinder49, 2020-03-12:
> This DAC doesn't specify the max signal rise time, but it is likely still sensitive to it (any ideas @gkasprow about what we should aim for?).

Footnote 2 on page 6 of the data sheet states: 

> 2 All input signals are specified with tR = tF = 1 ns/V and timed from a voltage level of (VINL + VINH)/2

This isn't a maximum, but our rise times are well within spec.

[4 more replies]

closed 2020-05-20 by gkasprow