sinara-hw/Fastino #64: Re-Route DAC Digital Traces

pathfinder49, 2020-02-21:
Currently the digital traces are routed close to analogue vias, causing crosstalk (see #62). It seems like there is enough space to avoid this with more careful routing.
[image]

pathfinder49, 2020-02-21:
I'll have a go at doing this next week.

gkasprow, 2020-02-22:
True, but make sure there is no excessive coupling between them. It may be worth adding another layer if that helps.

gkasprow, 2020-02-22:
Before you do this, perform a simple simulation. Model the trace-via capacitance and add it to the AFE model. If the simulation is consistent with measurements, it makes sense to modify the PCB. If not, we have to find a better explanation. The trace-via capacitances are very, very low and I'm surprised that you are able to measure their influence.

pathfinder49, 2020-02-25:
I've made a quick estimate based of these [trace-via coupling measurements](http://www.sigcon.com/Pubs/news/8_01.htm). 
* Digital trace-via separation on Fastiono is 5 mil.
* Via diameter is 15 mil
* digital trace width is 5 mil
Rounding up to the next tabulated trace-via spacing, rescaling and accouting for PCB permitivity=4, I get a trace-via capacitance of 4e-3 pF. (This is a somewhat coarse assumption, though should be in the right ball park)

I assume:
* SCLK is a 3.3 V square wave @50 MHz 
  * turend on for 16 cycles each DAC update (at 2.55 MHz)
  * From fourier series this gives a 2.55 MHz SCLK component of 1.05 V
* Effective circuit is a 4e-3 pF capacitor (trace-via) in series with a 50 Ohm resistor (spectum analyser).
  * This neglects the CMC. This should only give a factor of 2 as noise only couples to one of the traces.

With these assumptions I expect a 3.4 uV (-49 dBmV) spur @2.55 MHz. This is consistent with my observations.

pathfinder49, 2020-02-26:
> 
> 
> True, but make sure there is no excessive coupling between them. It may be worth adding another layer if that helps.

@gkasprow What would you consider excessive copling between digital traces? I think it'll be quite hard to avoid the traces running parallel to each other. Are there specific metrics I should achieve or good rules of thumb?

gkasprow, 2020-02-26:
Just run SI simulation after modifications. This is the reason I added another layer and routed some traces on it.

gkasprow, 2020-02-26:
A good approach is to use 5 trace widths as a clearance. And try not to run the traces in parallel over entire length

pathfinder49, 2020-02-26:
With the current number of layouts and via positions, it's not straightgforard to route the digital traces with the suggested clearance. (One trace width spacing with paralel routing for significant fratiions of the trace seems more doable.)

hartytp, 2020-02-27:
@gkasprow What do you think about [these](https://github.com/sinara-hw/Fastino/issues/64#issuecomment-590648321) calculations? They seem to back up the idea that the vias are an issue.

> Just run SI simulation after modifications. This is the reason I added another layer and routed some traces on it.

How did you run the SI simulation?

> A good approach is to use 5 trace widths as a clearance

Is it worth setting this up as a design rule for the digital traces?

> With the current number of layouts and via positions, it's not straightgforard to route the digital traces with the suggested clearance. (One trace width spacing with paralel routing for significant fratiions of the trace seems more doable.)

If we remove the CMCs do we still need the extra layers? If so, let's go for it unless there are any objections (@jordens )

gkasprow, 2020-02-27:
I used Hyperlynx to do SI simulations. Altium has also such features.
Yes, design rule is the best approach
CMCs will free some space for correct series termination.
I added parallel termination because the signals after SI simulation simply looked unacceptable. I shared the results as an issue.

gkasprow, 2020-02-27:
I think we need at least one more GND layer.

closed 2020-05-20 by gkasprow