sinara-hw/Fastino #65: Remove CMCs?

pathfinder49, 2020-02-26:
Using CMCs can offer improved noise performance in some configurations (see #56). 

However:
* The CMC location on Fastino is not ideal. Fasino has capacitive noise coupling mechanisms injecting current noise on the output side of the CMC. Using the CMCs on Fastino results in worse digital crosstalk (#62). 
* The digital traces should be rerouted to reduce digital crosstalk (#64). There are space challenges for routing the digital traces with sufficient inter-trace clearance. Removing the CMCs would allow the analogue vias to be moved closer to the op-amp. This would yield more space for digital trace routing.

I would suggest removing the CMCs from the Fastino design. As the CMCs are desirable in some configurations (#56), we may want to design an expansion board with CMCs. This would plug into the DAC headers on Fastino and sit in the adjacent slot. This would eliminate both the above mentioned drwbacks of the Fastino CMCs.

gkasprow, 2020-02-26:
Good idea with removing CMCs and makins space. In this way we would also add series termination close to the FPGA.

hartytp, 2020-02-27:
:+1: @pathfinder49 

Removing the CMCs would also remove many of the most problematic vias.

Assuming there are no objections (@jordens) I say let's go for it.

closed 2020-05-20 by gkasprow