sinara-hw/Fastino #66: Add More Layers?

pathfinder49, 2020-02-26:
Routing the digital traces with sufficient clearance to each other and the analogue vias is proving challenging. A possible solution for this would be to add (2?) more digital layers to the design.

pathfinder49, 2020-02-26:
@gkasprow How much cost would this add? Is this a route we want to go down?

gkasprow, 2020-02-26:
It's a 10% of PCB cost increase. Anyway, the PCB cost is not significant here.

hartytp, 2020-02-27:
@gkasprow since we want this to be a low noise design, it sounds like more layers would be a good idea if the cost increase is really that small compared with the overall board cost. As we're at 8, would that mean going to 12 layers? What stackup would you suggest?

@pathfinder49 can you update the stack up in Altium and post a screenshot here so we check we're all happy (or, better yet, push a release with more layers so Greg can check). You'll also need to check the design rules for the new layers...

hartytp, 2020-02-27:
@gkasprow would you want to use all the new layers for digital signals/ground or do you want to use some for analog signals/power/etc?

gkasprow, 2020-02-27:
When I was routing the board, I was missing one GND layer. So 10 layers would be fine. You can generate a realistic stack up on Brandner.ee website. We don't have many impedance controlled traces so it is not critical.
- add one GND layer to not share L3 as a ground plane and supply
- add the second GND between L5 and L6
- remove CMCs and parallel termination, make space for series termination close to the FPGA. 
We do not need more signal layers, there is enough space to separate the digital traces.
Some GND layers can be used as routing layers close to the FPGA.

pathfinder49, 2020-03-02:
@gkasprow I've got several questions on this.
1.  Why is having shared power and ground in L3 a problem? If so why is there ground in L3? L4 is a ground plane (see image below).
2. From some initial reading and Brandner.ee it looks like core and prepreg dielectrics should alternate. However, the current stackup does not adhere to this. Could you please explain this?
3. When adding layers, what is an adequate choice of dielectric type, material and thickness?
[image]

pathfinder49, 2020-03-17:
Aftyer removing the CMCs, I've managed to make do without extra layers.

closed 2020-03-17 by pathfinder49