sinara-hw/Fastino #67: Reduce ground clearance around vias?

pathfinder49, 2020-02-26:
Fastino 1.0 has significant digital to analogue crosstalk caused by trace-via coupling (#62). Reducing the via-ground plance may help shield this crosstalk. This should mostly effect traces where there currently is no ground plane between relevand vias and traces.

pathfinder49, 2020-02-26:
Currently the ground-via clearance is 20 mil. @gkasprow Is a reduction feasible from the production side?

gkasprow, 2020-02-26:
what do you mean by ground-via clearance? do you mean anti-via diameter?

gkasprow, 2020-02-26:
the standard distance is 0.2mm between via and the power plane.

gkasprow, 2020-02-26:
OK. the ground plane distance is 0.2mm but the polygon plane distance is 20mils. This can be reduced to 0.2mm easily

hartytp, 2020-02-27:
@pathfinder49 Can you keep a change log of all the changes you make to Fastino in your next commit?

It might be worth you fixing the "easy"/uncontroversial things (PSU grounding, design rules/clearance fixes, etc) before you get stuck into the routing.

closed 2020-05-20 by gkasprow