sinara-hw/Fastino #69: IC13 (LT3045EDD#PBF) out of spec

pathfinder49, 2020-03-02:
Following up on #54, I noticed that the P12V0A  supply (IC13, LT3045EDD#PBF) is also out of spec. The data-sheet requires a 10 uF output capacitance, but the same inssuficient X5R capacitor as for the N12V0A supply is used.

I gather that there may have been some confusion with the LT3042 supply. This requires only half the output capacitance. It might be worthwile checking this hasen't happended slipped into other sinara-hw projects.

hartytp, 2020-03-02:
Yeah. We should do a complete review of all the Sinara cards with LT3045 and check the output capacitance is sensible.

hartytp, 2020-03-02:
Based on the experience with this design, a single 22uF capacitor may not be sufficient to ensure stability reliable, depending on the regulator output voltage and capacitor voltage rating (e.g. a 16V cap at 3V3 is likely fine, but not at 13V). We should check all occurrences in Sinara and add extra capacitors anywhere that looks like it might be marginal.

@gkasprow is it easy for you to search through the Sinara designs and look for all occurrences of the LT3045? If so, would you mind doing that and opening issues anywhere where it seems appropriate (it may be that in all other cases the capacitors are fine as they are due to lower output voltages, but we should confirm this)? Once that's done we can close this issue.

hartytp, 2020-03-12:
This applies to Urukul as well I think...

hartytp, 2020-03-21:
@gkasprow I checked all Sinara projects for LT304x and filled issues where appropriate.

IIRC there was a negative LDO in the same family that use used in one or two designs. Can you remember what the part number was and check where we used it?

gkasprow, 2020-03-21:
it is LT3094, it was used in Fastino, but there is already 22uF

hartytp, 2020-03-21:
Didn’t we use that in another design as well? I thought I saw it somewhere else...

Anyway as we know 22uF is not enough once voltage derating is taken into account since it requires a min of 10uF. So we need more capacitance there (@pathfinder49 fixed this on fastino already but not on any other designs)

gkasprow, 2020-05-20:
I added another 33uF for IC21

closed 2020-05-20 by gkasprow