sinara-hw/Fastino #70: Analogue Crossalk

pathfinder49, 2020-03-04:
The output voltages of different Fastino channels have very little cross-talk.

# Method
I wrote channel 16 (central on the board) to a fixed voltage. I then used a 6 digit multimeter (Hewlett Packard 34401A) to look for changes in the channel 16 voltage when toggling all other channels between +-10 V

# Result
* Setting channel 16 to mid-range (0 V) I saw a ~1 uV difference. (0.1 uV least significant digit) 
* Setting channel 16 to +-10 V (0 V) I saw a 10 uV difference. (10 uV least significant digit)

hartytp, 2020-03-04:
Nice!

So the second measurement (output at +-10V) tells you about reference cross-talk (scales with output word) while the first tells you about offsets.

AFAICT the reference cross-talk is a null result (nothing within the dmm resolution, although we do have a 7.5 digit one somewhere so if you have time it might be nice to get a more accurate reading).

There is a small offset measured there. 1uV on 20V is 0.05ppm. That’s eligible e.g. compared with the reference stability. 

So overall the conclusion is that there is no significant dc cross talk *with channels driving a high-Z load* like a capacitor

hartytp, 2020-03-04:
Out of curiosity did you spot check a few other channels to check there are none that are significantly worse?

pathfinder49, 2020-03-06:
I've repeated the measurements with the 7.5 digit multimeter and on some more channels. 

| fixed channel | channel set to -10 V (reads to 1uV) | channel set to +10 V (reads to 1uV) | channel set to 0 V (reads to 1uV)|
| --- | --- | --- | --- |
16 | 10 uV | 10 uV | no sig. difference
24 | 20 uV | 25 uV | no sig. difference
1 | no sig. difference | no sig. difference | no sig. difference

hartytp, 2020-03-07:
> I've repeated the measurements with the 7.5 digit multimeter and on some more channels.

Cool! To confirm, what values are you using for the other channels? Just switching them between +-10V?

In your table you say that the resolution for 0V is 1uV. The DMM can do 0.1uV. Was that a typo in your table or did you not adjust the DMM to give the best possible resolution?

pathfinder49, 2020-03-07:
That was a coppy paste error. Fixed.

hartytp, 2020-03-07:
Based on the (admittedly somewhat small) data set you posted there, it looks like:
1. The cross-talk is small (~0.5ppm/0.05LSB FS worst-case). Assuming that's really representative, I suspect it's lost within the noise of thermal fluctuations in the reference etc
2. The cross-talk is predominantly on the reference signal since it scales with the output voltage
3. There is quite a bit of channel-channel variation in the noise. By eye, it looks like this may well just correspond to the distance from the reference: both ch1 and ch16 are driven by IC14; ch1 is close to the reference and has no noticeable cross-talk, while ch16 is a long way from it and has worse cross-talk.

@gkasprow I don't want to spend a long time on this, but I am curious about one thing...Given the DAC architecture, can we produce a kelvin connection by running a reference return line from the ground side of RN5C back to a point near the ref buffer? (e.g. to test, we could lift pin 6 of RN5 up and solder an air wire to connect this to GND close to the ref buffer. Then add a 100nF decoupling capacitor between the REF and RN5C close to the DAC to provide a low impedance noise bypass). Or, is that a silly idea?

hartytp, 2020-03-07:
Thanks for confirming @pathfinder49 !

pathfinder49, 2020-03-07:
> cross-talk is predominantly on the reference signal since it scales with the output voltage

The DAC is setup for unipolar operation. I would expect reference noise to be absent at -10 V and maximal at 10 V. Couldn't the observed behaviour result from the P/N12V0A supply?

hartytp, 2020-03-07:
> The DAC is setup for unipolar operation. I would expect reference noise to be absent at -10 V and maximal at 10 V. Couldn't the observed behaviour result from the P/N12V0A supply?

I don't think that's correct. The DAC is just a restive divider on the reference, but so are the OpAmp feedback resistors. i.e. at 0V output, the contributions from the DAC resistors and feedback resistors balance out to give no net dependence on the reference voltage.

Put it another way, if you vary the reference voltage, the -10V will change, so that obviously is dependent on the ref voltage and hence on any differences between the reference ground potential and the ground potential at the DAC...

hartytp, 2020-03-07:
Anyway, to be clear, I'm not suggesting that we should necessarily Kelvin connect the reference (although if it turns out to be easy in terms of layout, then it may be worthwhile) since these effects are currently tiny. This is mainly me indulging my curiosity...

hartytp, 2020-03-07:
Having said that I suspect that my previous description is incorrect and that there is no easy way to achieve a kelvin connection.

I really wouldn’t expect a dependence on the p/n 13V supplies. The opamps dc part should be very high and the current draw from those supplies isn’t changing significantly (since there is no dc load).

Do the mats and tell me if I’m wrong though:)

There will also be an effect due to board temp change but I’d expect that to be negligible —10uV is large!

hartytp, 2020-03-07:
@pathfinder49 youve done a great job of characterising this board. Let’s suggest one final measurement and then move to finishing the layout. The last measurement that I think would be useful would be an ac cross talk. Say a 200kHz square wave on all channels apart from one and then look at that channel on the spectrum analyser

hartytp, 2020-03-07:
(Actually if you have time it would also be lovely to see the spectrum as you take one channel back and forwards across a major carry. Then I really think we’ve seen everything we need!)

pathfinder49, 2020-03-09:
>  Say a 200kHz square wave on all channels apart from one and then look at that channel on the spectrum analyser

The next best thing I can do until the wider bus is implemented was a +-10 V, ~25 kHz square wave on all other channels. AC crosstalk is readily apparent on a scope:
[image]
The 25kHz spur power is -30-76 dBmV.

hartytp, 2020-03-09:
@gkasprow how is the agnd S pin intended to be used for these DACs?

gkasprow, 2020-03-09:
@pathfinder49 are you absolutely sure that this crosstalk is not caused by scope probes? Try to short the tip with the GND to make sure.

hartytp, 2020-03-09:
NB it could also be cross-talk in the BNC<->IDC board and cable (@jordens posted some measurements on that a while back that we should dig out).

hartytp, 2020-03-09:
https://github.com/sinara-hw/Zotino/issues/21#issuecomment-376119235

Those results aren't directly comparable (a) because @jordens used a 2m cable (screened, twisted pair) and (b) because he had a slightly different configuration (source/analyzer impedance etc). 

By eye, I'd guess the bulk of what you see is at a few hundred kHz (associated with the edge rise time on that waveform). For that frequency range, @jordens measured a -50dBc cross-talk. So, unless I've mis-understood the data @jordens posted, that suggests that in a practical configuration (with a cable going from Fastino to the trap) the cable cross-talk will dominate. Which is what one wants...

pathfinder49, 2020-03-10:
> 
> 
> @pathfinder49 are you absolutely sure that this crosstalk is not caused by scope probes? Try to short the tip with the GND to make sure.

This was measured with a BNC cable from the BNC<->IDC board.

hartytp, 2020-03-29:
@pathfinder49 can you create a layout review issue please? @gkasprow ping...could you perform a review of @pathfinder49's new layout / PR?