sinara-hw/Fastino #72: Review V1.1

pathfinder49, 2020-03-30:
Review chasnges for V1.1 from [this PR](https://github.com/sinara-hw/Fastino/pull/71).

Specific changes requiring attention:

- [ ] DAC-SPI signal integrity
- [ ] DAC-SPI series termination
- [ ] DAC-SPI parallel termination (DNP)
- [ ] P3V3_MP
- [ ] En_AFE_PWR pull down
- [ ] i2c dependencies
- [ ] IC3 grounding
- [ ] P/N12V0A capacitors
- [ ] DAC channel layout
- [ ] component clearances

gkasprow, 2020-04-01:
- [ ] I would connect pin 8 of IC3 to GND instead of pin 9. It's a subtle difference.
- [ ] I wouldn't pour the GND on the top and bottom layers. There is a subtle difference - it's easier to control how circuits are connected to GND. With copper pour, one can easily create parasitic GND connections between neighboring circuits. This also makes the crosstalk lower. Take into account that vias have inductances and resistances. 
- termination for 2cm traces do not make much sense but let's treat it as a lowpass filter and we want to have identical circuits for all channels
- [ ] C106_x should have a dedicated via to GND. Now it shares with power supply caps which will affect the PSRR
- use tented vias. The only exception is vias in the pad where it must be left open on both sides to not create acid pockets.

gkasprow, 2020-04-01:
- [ ] I would supply the IC6 and JP1 from P3V3. The IC17 is used to disconnect the I2C bridge from the P3V3 power domain.  When P3V3 is absent, the FPGA will pull all SPI lines to 0.7V.

gkasprow, 2020-04-01:
- [ ] the supply of the DAC is not optimal. 
[image]
The trace should follow the route: source -> inductor/resistor -> bulk cap -> ceramic cap -> pin
The critical is low impedance of the ceramic cap to GND. In this case the path from this capacitor goes to the neighboring channel.  Now the caps are far away from the IC. The original design did not have such problem
[image]
This can be  improved, for example
[image]

gkasprow, 2020-04-01:
- [ ] reference decoupling is critical. Now the decoupling capacitor is connected between the REF pin and GND of neighboring channels. I placed the decoupling cap between REF and GND on purpose
- [ ] reference bulk capacitor shares ground with a decoupling cap.

gkasprow, 2020-04-01:
Here are a few images from my teaching materials that explain what I meant. They come from random app notes
[image]
[image]
[image]
[image]
[image]
[image]

gkasprow, 2020-04-01:
- [ ] the ceramic caps are too far away from +/-12V pins of the opamps. It is not that important as in case of the DACs.
- [ ] LDAC is too close to high impedance opamp input
[image]
this can be fixed in such way
[image]

gkasprow, 2020-04-01:
-[ ] we can get rid of R153 ane make space for correct VREF decoupling

gkasprow, 2020-04-01:
- [ ] increase the clearance between the traces and copper pour because of massive antennas/resonators that cause additional coupling between tracks
[image]

pathfinder49, 2020-04-04:
> I would connect pin 8 of IC3 to GND instead of pin 9. It's a subtle difference.

I went with pin 9 as I tested that configuration. Is there a strong reason to go with pin 8 instead?

> reference decoupling is critical. Now the decoupling capacitor is connected between the REF pin and GND of neighboring channels. I placed the decoupling cap between REF and GND on purpose

Are you refering to the top and bottom GND polygon here?

pathfinder49, 2020-04-06:
@gkasprow I've pushed an updates addressing your points except the IC3 pin. Could you please take another look?

pathfinder49, 2020-04-14:
@gkasprow Do you know when you'll have a chance to look at this?

gkasprow, 2020-04-21:
@pathfinder49 next time pls write an email directly to me.

gkasprow, 2020-04-21:
@pathfinder49 where is the PCB? I cloned your repo and there is the old version..

pathfinder49, 2020-04-21:
It's on the branch "fixes" (apologies for the poor naming). https://github.com/pathfinder49/Fastino/tree/fixes

gkasprow, 2020-04-21:
I think it is good enough. Every project can be done better, but here the main goal is high channel isolation. Let's produce it;

closed 2020-05-20 by gkasprow

gkasprow, 2020-05-20:
I changed the rev_ID from 1 to 2

pathfinder49, 2020-05-21:
@gkasprow If you merge PRs through the webinterface, the linked issues can be closed automatically :wink: