sinara-hw/Fastino #75: V1.1 Initial Testing

pathfinder49, 2020-08-20:
I've finally got my hands on a V1.1 board and am running it through some rudamentary tests.

# Output Voltages
I've hooked up Fastino to an IDC-BNC adapter board. I then wrote all DAC channels to +9.5 V and measured the output voltages using a DMM (10 mV precision). All Channels read 9.5 V within the DMM precision.

# Power rails
I measured the rail volatges on the test points using a DMM.

*  P12V0A: fine
*  N12V0A: fine
*  P1V2: fine
*  REF_P5V0a-d: fine (reads 5.000 V,  1 mV precision)
*  P2V5: fine
*  P12V0: fine
*  P3V3: fine
*  P3V3MP: fine
*  P5V4A: fine

pathfinder49, 2020-08-25:
# Digital crosstalk
I've spot-checked some channels for digital cross-talk. I'm updating a single channel at the 2.55 MS/s. The channel is repeatedly written to -9.5 V (where -10 V would be the 0x0000 word). It is worth noting that the board I'm testing has the parallel termination populated. #62 suggests that the parallel termination is responsible for digital cross-talk to nearby DACs. This might be improved by desolating the parallel termination/using the variant that has the parallel termination de-populated.

Spur intensities are given in dBmV of the spur peak. The analyser noise floor is -90 dBmV.
## Channels picked for location on board and their trace routing
With the selected channel updating, all other channels were checked for a 2.55 MS/s spur.

From Channel 21 updates: no spurs seen
From Channel 28 updates: 
*  -80 dBmV spur in ch30 (this looks like the parallel termination cross-talk from #62)
*  -62 dBmV spur in ch8 (I don't currently have a model for this spur.)

## Channel 4 (comparison to #62)
I only measured channels 0-15 for ch4 updates as this contains all previously observed spurs for channel 4 and the related digital traces are well away from channels 16-32
*  -79 dBmV spur in ch6 (This looks to be the parallel termination mechanism mentioned above.)

# Conclusion
With the new layout, there seems to be less cross-talk to channels. The strongest spurs I fund during spot-checks are much weaker than those previously seen. However, I found a significant spur. This seems worth investigating further.

pathfinder49, 2020-08-25:
On further investigation, there is a major digital cross-talk issue with channel 8. Details are presented in #76. Essentially this renders channel 8 very susceptible to digital cross-talk. Other channels appear to be effected to a lesser degree.

hartytp, 2020-08-26:
Thanks @pathfinder49 

Could you clarify a few things here to make sure I understand correctly:
- Am I right in thinking that this board has both parallel and series termination (see below)? Doesn't that cause major SI issues? Was this the intention, or was the plan to remove the parallel termination and then check the SI?

[image]
[image]

- What was the most comparable data set you had with the old hardware revision? I'd like to do a like-for-like comparison to see how the results compare.

hartytp, 2020-08-26:
One other question: if we discount the channels that are near the termination resistors, am I right in thinking that the worst-case digital spur you measured was ~-80dBmV (100nV RMS / -157dBFs) at 2.55MHz (so should be easy to get another factor of ~10-100 from the trap filters) and that you believe (but haven't verified) that this is due to return currents from the parallel termination at the DAC? Obviously it would be nice to eliminate that by removing the parallel termination resistor (assuming that neither SI or issues around the series termination emissions prevent it), but I'd be okay living with that if we have to.

If we decide to remove the series termination resistors, we should repeat this measurement to verify that the spurs don't get significantly worse (due to the fact that the series resistors were attenuating the high-frequency components of the digital lines)

pathfinder49, 2020-08-26:
To confirm my interpretation that the -80 dBmV spurs seen in adjacent channels are due to the parallel termination, I desoldered the parallel termination capacitors of channel 28 and channel 4 and measured the effect on the spurs.

Removing the parallel termination of channel 4 does indeed remove the -79 dBmV spur seen in channel 6. This also matches the V1.0 measurements presented in #62. 

Removing the parallel termination of channel 28 does **NOT** influence the -80 dBmV spur seen in channel 30. 

This suggests that the parallel termination is the cause of some (but not all) -80 dBmV spurs in adjacent channels. Given the difficulties arising from the series termination (#76), it is worth revisiting if we want to use series or parallel termination.

pathfinder49, 2020-08-27:
>     * What was the most comparable data set you had with the old hardware revision? I'd like to do a like-for-like comparison to see how the results compare

The channel 4 measurments should be directly comparable to the measurements for channel 4 in #62.

>     * Was this the intention, or was the plan to remove the parallel termination and then check the SI

Indeed. There are two 1.1 variants. (One with and one without the parallel termination capacitors populated.)

hartytp, 2020-08-31:
@pathfinder49 I think we've now done all the testing we planned on this revision, haven't we. Shall we close this?

hartytp, 2020-08-31:
Actually, no, let's keep this open until we've made a decision about termination for the digital lines and done some SI spot checks with a low capacitance probe.

pathfinder49, 2020-09-01:
I've looked at the digital SI with only one channel receiving digital updates. The channel update rate was 2.55 MHz.

Ideally, I would have liked to include clock jitter in this measurement by triggering off the CLK. However, I have no good way to get two high-speed probes onto the board simultaneously.

# Series termination 
I measured SI with series termination on CH27 (ch28 in altium) as this has the longest digital traces. These traces are with 30 - 60 s peristance.

Parallel termination capacitors were removed for this measurement.
| CLK | DIN |
| --- | --- |
[image] | [image]

# Parallel termination
I measured SI with series termination on CH30 (ch31 in altium). This channel also has a long digital trace.

The series termination resistors were replaced with a short circuit for this measurement.
| CLK | DIN |
| --- | --- |
[image] | [image]

hartytp, 2020-09-02:
@pathfinder49 thanks for performing those measurements.

Am I right in thinking:
- series termination works nicely: the rise time is comparable to the current parallel termination, but without the overshoot/undershoot measured at the DAC terminal (the parallel termination looks like the overshoot is still above the 3V3 abs max rating, so the termination would need tweaking anyway). Based on that, it appears that the SI is good for this channel with series termition
- the series termination you're using there is probably actually a little aggressive on this channel; we have no overshoot at all and RC filtered edges.
- the series termination you measured had the longest trace, which is expected to give the lowest undershoot (lowest capacitance so least aggressive RC filter). Could you also make a measurement of the shortest trace? If the trace length difference is large we may need to use a couple of different termination resistor values to get good waveforms on all channels (or it may be that we're fine with things as they are, it just needs a spot check on a few channels with different routing lengths to decide)

So, it sounds like the conclusion (pending checks on a few other channels) is that the series termination works really nicely and gives a better SFDR so long as we can sort out the placement/routing of the termination resistors and associated traces + vias.

Is that all correct?

pathfinder49, 2020-09-02:
After discussion, @hartytp and I agree these points are not of concern.

*  The Parallel termination is within the DAC spec. (VDD = 5.4 V)
[image]

*  As per this measurement series (from v1.0), we do not want to change the series termination resistance.
https://github.com/sinara-hw/Fastino/issues/63#issuecomment-597647786

*  Due to the reduced length of the shorter traces, these don't need termination as much. Further, the DAC capacitance is 10 pF, which is comparable to that of the longest traces. The RC filter cut-off will therefore not change that much.

hartytp, 2020-09-16:
We've completed our testing of v1.1.

closed 2020-09-16 by hartytp