sinara-hw/Fastino #85: V1.2 Initial Testing (and kasli v2)

pathfinder49, 2021-02-08:
# Apparatus
*  Kasli v2.0
    * Powered by a class II supply with a CMC to cut the AC-ground-loop.
    * Not touching any grounded surfaces.
*  Fastino v1.2 powered of EEM1 of the Kasli
    * Taped onto table well away from Kasli. This shouldn't make a big difference as long as Fastino is a few HP away from Kasli (#56).
*  Agilent E4405B spectrum analyser (50 Ohm) 
    *  Connected via IDC-BNC breakout board with CMCs
    *  Use a 100 kHz DC block to protect the analyser for DC voltages
*  GND-strap from Fastino front pannel to Spectrum analyser ground

**In the following all references to channel numbers are by the channel number printed on the board (same as in artiq code). This number must be converted to Altium channels by adding one to the channel number.** (Ex: channel 28 is channel 29 in Altium)

# Non-Fastino Update Spurs
Looking at channel 24-31 this configuration was verified to produce a "spur free" spectrum in the 100 kHz -> 5 MHz range. For further investigation of "spur free" configurations please see #56.

# Power rails
All power rails were within tolerance when measured with a DMM.

# Fastino Digital to Analogue Cross-Talk
Fastino 1.1 saw significant digital to analogue cross-talk to channel 8. This was as large as -33 dBmV (channel 17 to channel 8) #76. Further, some unexplained -80 dBmV spurs were observed #75. The primary aim of revision 1.2 was to reduce the digital to analogue cross-talk to channel 8. In the following I present measurements that is comparable to the measurements in #76 and #75.

## Configuration
*  As described in apparatus (top)
* An individual Fastino channel "agressor" was written to -9.5 V at 2.55 MS/s. (where -10 V is 0x0000)
* The spectrum analyzer was used to measure the resulting 2.55 MHz analogue output spur in the "victim channel". The resulting spurs are narrower than the 10 Hz analyser bandwidth.

## Measurements

Keeping channel 28 as the agressor, crosstalk to all channels was measured: (in dBmV, -85 is noise floor) 
- Ch 0: none 
- Ch1: none 
- Ch2: none 
- Ch3: none 
- Ch4 none 
- Ch5 none 
- Ch6 none 
- Ch7 none 
- Ch8: none 
- Ch9: none 
- Ch10: none 
- Ch11: -85 
- Ch12: none 
- Ch13 none 
- Ch14: -83 
- Ch15: none 
- Ch16 none 
- Ch17 -84 
- Ch18 none 
- Ch19 none 
- Ch20 -83 
- Ch21: none 
- Ch22: none 
- Ch23: -84 
- Ch24: none 
- Ch25: none 
- Ch26: none 
- Ch27: none 
- Ch 28 (self): -75 
- Ch 29: none 
- Ch 30: -80 
- Ch 31: none 

Keeping channel 8 as the victim, crosstalk from all channels was measured: (in dBmV, -85 is noise floor)

- Ch0: none 
- Ch1: none 
- Ch2: none 
- Ch3: none 
- Ch4: none 
- Ch5: none 
- Ch6: none 
- Ch7: none 
- Ch8 (self): -75 
- Ch9: none 
- Ch10: none 
- Ch11: -80 
- Ch12: none 
- Ch13: none 
- Ch14: -82 
- Ch15: none 
- Ch16: none 
- Ch17: -76 
- Ch18: none 
- Ch19: -81 
- Ch20: -65 
- Ch21: none 
- Ch22: -75 
- Ch23: none 
- Ch24: none 
- Ch25: none 
- Ch26: none 
- Ch27: none 
- Ch28: none 
- Ch29: none 
- Ch30: -74 
- Ch31: none 

Performing a spot check with ch20 aggressor to victim channel 11, no spur is seen.

The worst digital to analogue cross-talk spur was measured as -65 dBmV. This is from aggressor ch20 to victim ch8. It is not immediately clear what the coupling mechanism for the ch20 to ch8 is. Regardless, this cross-talk is improved from -55 dBmV in Fastino v1.1

From the schematics, the ch20 to ch8 cross-talk is likely to be a worst case (see below).

## Digital to Analogue Summary
Digital to analogue cross-talk in V1.2 is improved compared to all previous revisions. The worst-case digital to analogue cross-talk in v1.2 is over 30 dB better than in v1.1. This is somewhat worse than hoped for V1.2, but seems tolerable.

Routing of ch20 digital traces (highlighted) around ch8 (bottom) and ch11 (top).
[image]

hartytp, 2021-02-08:
> Powered by a class II supply with a CMC to cut the AC-ground-loop.

Did the new layout for Kasli help? Is the CMC still needed?

> Looking at channel 24-31 this configuration was verified to produce a "spur free" spectrum in the 100 kHz -> 5 MHz range. For further investigation of "spur free" configurations please see #56.

🎆 

> The worst digital to analogue cross-talk spur was measured as -65 dBmV.

> Digital to analogue cross-talk in V1.2 is improved compared to all previous revisions. The worst-case digital to analogue cross-talk in v1.2 is over 30 dB better than in v1.1. This is somewhat worse than hoped for V1.2, but seems tolerable.

That's pretty impressive! Good work.

To put this in perspective, that's a 560nV RMS spur on a +-10V output. I would generally expect trap filters to add a bit to this as well.

I'm happy to leave this investigation here; it feels like we're solidly in the realm of diminishing returns and there isn't (as far as I'm aware of) a physics/user reason to need to push this further. Nice job all!

pathfinder49, 2021-02-09:
From skimming the [publication on a similar design by Mainz](https://avs.scitation.org/doi/pdf/10.1116/1.5126186), I've compiled a quick comparrison. Well done all 👍 
### Output range
*  Mainz: +-40 V
*  Fastino: +-10 V
### Output channel number (per module)
*  Mainz: 16
*  Fastino: 32
### Update Rate (per cahnnel)
*  Mainz: 1.7 MS/s? (Dac capable of 2.55 MS/s)
*  Fastino: 2.55 MS/s
### Boradband noise @1 MHz
*  Mainz: ~220 nV/rt(Hz)
*  Fastino: ~1.4 nV/rt(Hz) (measurement: #56, simulation: #4)
    *  (56 nV/rt(Hz) @ 100 kHz)
### Noise Spurs
*  Mainz: Many spurs are visible, especially when the dac is updating. These are quite significantly above the noise floor. However, the measurement bandwidth is not given. Therefore I can't convert to a spur power. Assuming 10 Hz measurement bandwidth digital update spurs are ~30 uVrms @1.7 MHz.
*  Fastino: No spurs without digital updates in the 100 kHz to 5 MHz range. Digital update spurs at 2.55 MHz are ~560 nVrms @2.55 MHz.

gkasprow, 2021-02-09:
Great work!

pathfinder49, 2021-02-11:
> Did the new layout for Kasli help? Is the CMC still needed?

I'll investigate the impact of the new kasli layout later and will open a new issue once I can make conclusions (either here or sinara-hw/Kasli)

closed 2021-02-11 by pathfinder49