sinara-hw/Fastino #87: Potential new DAC

dtcallcock, 2021-12-14:
[AD3542R](https://www.analog.com/en/products/ad3542r.html#product-overview)
2ch, 16-bit
15/10 MSPS
Dual SPI with DDR

I haven't done a careful side-by-side comparison.  But could potentially offer a really nice 4-5x speed bump for Fastino if found suitable.  

Chip still in pre-release...

gkasprow, 2021-12-14:
Yes, the ADI representative was talking that they are aware of the lack of precise 16bit DACs running at ~10MHz rate.

dtcallcock, 2021-12-14:
They promised a product so perhaps this is it.

jordens, 2021-12-14:
It's only 7.5 or 5 MS/s if you want to use both channels.

gkasprow, 2021-12-14:
Right. The clock can go up to 60MHz in the dual data line, 2 Byte,  DDR mode which gives 240Mbit/s data rate and 15MS/s...

dtcallcock, 2022-02-10:
Another new 'fast' SPI DAC: [AD3552R](https://www.analog.com/en/products/ad3552r.html#product-overview)

Unlike the AD3542R, this one seems to have been released.

DDR quad SPI allows 33 MUPS in 'fast' mode and 22 MUPS in 'precision' mode.  The clock can go up to 66MHz on the quad data line, 2 Byte, DDR mode which gives 528Mbit/s data rate and 33MS/s.  It's a 2ch chip so obviously halve that if running both channels.

Requires external transimpedence amps.

$40 compared to 2x $10 for pair of AD5542A, so twice the silicon cost but nothing crazy.

Noise seems a bit worse not specced the same way in datasheet so not obvious at a glance how much worse.  Needs checking.

Power consumption is about 200mW so 32ch on a Eurocard should be manageable.

dhslichter, 2022-02-11:
This is getting fast enough that one might do fast transport as well as sinusoidal modulations (e.g. for squeezing, mode-mode coupling, etc).  Would it make sense to look at trying to harmonize gateware/user API between Ultra-Fastino and Shuttler, to reduce wheel reinvention since probably similar signals and parameterizations would be used?

dtcallcock, 2022-02-11:
> Ultra-Fastino

Surely it should be called '2Fastino'

dhslichter, 2022-02-11:
I guess maybe the thing to do would be to have "2Fastino" (2Furiousino?) be a variant on an FMC daughtercard, and run it from the EEM FMC Carrier in the same way as Shuttler.  That would cut down on the channel count but still 16 channels is probably doable.

gkasprow, 2022-02-11:
FMC Shuttler has 125MS/s DACs already, but they are 14-bit ones. 
So we have two options:
- FMC DAC + external 16-channel AFE + ARTIX carrier board
- Updated Fastino with faster DAC and bigger, low-cost FPGA

dhslichter, 2022-02-11:
I'm not trying to create more work here, so maybe this isn't a great idea.  And I want to keep Shuttler as is because the higher sample rate is important for some of our applications.  I was just thinking out loud about, if one were to make a Fastino variant with some of these faster SPI DACs, if it wouldn't make sense to have the design share more platform features with Shuttler (and its modular FMC nature), and thus be able to harness the gateware development for Shuttler as well.

Not trying to rock the boat here!

gkasprow, 2022-02-12:
The intention of my comment was a bit different. To decide which way to go, we must first answer the question how much FPGA resources we would need. With EEM FMC carrier we have 100k LC per 16 channels.
With ECP5 FPGA and updated Fastino we have 85k LUT.  Lattice's LUT is not equal to Xilinx's LC of course.
The second question is what use cases do you see for this board that cannot be achieved with Fastino and Shuttler.

dtcallcock, 2022-02-13:
> The second question is what use cases do you see for this board that cannot be achieved with Fastino and Shuttler.

My thought was that this would be a straightforward replacement for Fastino, as it seems it would really improve its utility for fast-ish transport.  Probably best if big Fastino users like @hartytp weigh in on that though.

gkasprow, 2022-07-31:
Adding interesting DAC do the list: [AD3552R](https://www.analog.com/en/products/ad3552r.html?ADICID=EMAL_WW_P628000_MIX-NL-PN_1125&deliveryName=DM25392#product-overview)

closed 2024-03-28 by gkasprow