sinara-hw/Fastino #92: SC18IS602B is EOL

MorganTL, 2023-10-17:
The manufacturer recommends using SC18IS606PWJ, but the pinout is different
[image]
[image]

gkasprow, 2024-03-28:
Both chips look nearly identical from the register/commands point of view. The only difference is the lack of a GPIO3/SS3 pin.
FUnction FD7h has different bit allocation
SC18IS602 chip
[image]
SC18IS606 chip
[image]

SC18IS606 chip has additional "Read Version: Function ID FEh"

We use GPIO3 as FPGA configuration reset.
@jordens can we live without one of GPIOs?
[image]

gkasprow, 2024-03-28:
I think we could merge SPI FLASH SSEL and FLASH_UP_EN (via inverter)

gkasprow, 2024-03-28:
A better idea is to control the switch with CRESET_B
Anyway, we keep resetting the FPGA configuration engine when talking to FLASH.

gkasprow, 2024-03-28:
such circuit could work
[image]