sinara-hw/Fastino #127: remove one reference

gkasprow, 2024-08-28:
Some boards have both references assembled. It's an error in the assembly variant.
When references have very similar voltage, Fastino works excellent; the noise of references might be lower.
However, when both references differ by more than a few mV, they start to fight against each other and overheat/oscillate.
Fastino may work properly for some time and, after minutes, produce strange effects.
The remedy is simple - remove the IC12
[image]

dnadlinger, 2024-08-28:
Is there any indication which board revisions produced by which manufacturer across which time period this affects?

jordens, 2024-08-28:
current master since 2022, e.g. several 2302020PW0020Axxxx by CTI

dnadlinger, 2024-08-28:
Thanks for the quick reply – to check, has this only occurred for boards with the ".1" patch release on the silkscreen (I'm having some Altium licensing issues here)? It would be good to get a consise user advisory note together for this.

A minor point: Isn't the LS8 pacakge supposed to be more stable, i.e. would it make sense to change the advice to remove IC11 if both are populated?

gkasprow, 2024-08-30:
One can also remove the second one, but it needs the NR pin to be disconnected, so the 0R resistor (R179) must be removed as well
Otherwise, the NR would be shorted to OUTF, and the reference may not like it
[image]

kaolpr, 2024-09-02:
For the record, there is no error in assembly variant. Default variant for the project is `STD` (see [PrjPCB](https://github.com/sinara-hw/Fastino/blob/45785f2edd1ef654d92918008e8e4980548a9eec/PCB/Fastino.PrjPCB#L45)). It is also selected as a variant for output jobs (see [OutJob](https://github.com/sinara-hw/Fastino/blob/45785f2edd1ef654d92918008e8e4980548a9eec/PCB/Fastino.OutJob#L21)). Having `STD` variant active makes IC12 DNP.
[image]

gkasprow, 2024-09-02:
I don't know about the assembly variant. There were issues with noise induced by parallel termination, and that's why we introduced serial one...

gkasprow, 2024-09-02:
https://github.com/sinara-hw/Fastino/issues/80